3dtof/voxelsdk #38: Example code request - calibrated point cloud

larrylisky, 2015-11-06:
Is there voxelsdk C++ example showing how to get calibration point cloud? The current example does not seem to provide calibrated point cloud (selecting the profile that has the proper calibration?).

hlprasu, 2015-11-06:
Could you please specify which device you are using? TintinCDK?

gadiyar, 2015-11-06:
I think it's common for all devices, right? The examples code (VoxelCLI or SimpleVoxelViewer for example) for users does not show how to set a profile. I think this is only available in the VoxelViewer code right now.

larrylisky, 2015-11-06:
Correct, we need an example that works for all devices, as long as the calibration profiles exist on the CDK hardware.

hlprasu, 2015-11-06:
If you have not created any new profiles, then pre-installed default
profile will be used.

If you have created a new profile which is made default then it will be
used in Voxel CLI or Simple Voxel Viewer.

From the question, I understand that the default profile is not calibrated
appropriately.

@Larry: Could you please the messages shown when you run VoxelCLI or
SimpleVoxelViewer?

larrylisky, 2015-11-23:
@hlprasu:  The question was referring to how to access pre-calibrated profile stored on the CDK.  After lens calibration, the MetrilusHWProfile is stored to the CDK and that's the one I want to use.  How do I load that profile by name (or by ID) during execution?

hlprasu, 2015-11-23:
In SDK, you can use depthCamera->setCameraProfile(id) to set the desired camera profile.

You can use depthCamera->getCameraProfileNames() to get a map object of all valid IDs to names. Note that IDs are unique and names need not be.

closed 2015-12-04 by hlprasu