3dtof/voxelsdk #39: install libpcl-all does not install libpcl-visualization.so

larrylisky, 2015-11-06:
I noticed in my latest Ubuntu install of pcl following the below command:

      sudo apt-get install libpcl-all

it does not install libpcl-visualization.so needed b voxelsdk.  I had to download pcl-1.7.2 directly from PCL  Github and build it to have PCL properly installed.  Please verify this and update the wiki.

hlprasu, 2015-11-06:
Which version Ubuntu are you using?

larrylisky, 2015-11-06:
14.04

larrylisky, 2015-11-06:
Also I think PCL master branch has moved onto version 1.8.

hlprasu, 2015-11-06:
I just checked again that libpcl-all should work fine. It in-turn installs
libpcl-visualization-1.7-dev which creates the required
libpcl_visualization.so.

Could you please check whether 'apt-get install libpcl-all' was successful
for your installation?

Is this issue for 32-bit/64-bit Intel/AMD architecture or for ARM
architecture?

larrylisky, 2015-11-06:
Him. Didn't work for me. Will try again. 

Larry

hlprasu, 2015-12-04:
Is this issue still there?

gadiyar, 2017-04-26:
Closing - no further action. Please reopen if this is still an issue.

closed 2017-04-26 by gadiyar

joerowelll, 2023-02-12:
I am getting the same problem, the apt version of pcl doesnt install visualization. Is there a way around this without building from source?

gadiyar, 2023-03-11:
Are you on a recent version of Ubuntu? This project is quite old and later PCL versions might have changed.

I would recommend building from source.

Also, do you have one of the depth cameras that this project supports? They're quite rare to come by these days.