3dtof/voxelsdk #42: Include files in 'ti3dtof/boards missing in the DEB installer package

larrylisky, 2015-11-17:
As result, I can't include TintinCDK specific header files.

hlprasu, 2015-11-17:
TIt is intentional considering that new board developers would typically
link to TI3DToF library and not board-specific libraries directly.

Is this a requirement? If so, I can do the necessary changes. But, note
that the board-specific libraries are installed in /usr/lib/voxel and not
/usr/lib. So, you'll have to set LD_LIBRARY_PATH to point to /usr/lib/voxel
when running programs which link to these libraries.

larrylisky, 2015-11-17:
I was  accessing those header files because I was trying to get setting 
of illum_power_percentage to work.

So far, no luck.

This is my function call:

     cout << "flag=" << depthCamera->set("illum_power_percentage", 20U) 
<< endl;

This is the error message.

     ERROR: DepthCamera: Could not set value 20 for parameter 
TintinCDKCamera(0::0451:9105::791070790962).illum_power_percentage
     flag=0

Any ideas?

Something in this function of ParameterTemplate class is failing...

   virtual bool set(const T &value)
   {
     if(_ioType == IO_READ_ONLY || !validate(value))
     {
       return false;
     }

     if(_programmer.setValue(*this, _toRawValue(value), _ioType == 
IO_WRITE_ONLY))
     {
       _value = value;
       return true;
     }
     else
       return false;
   }

-Larry

hlprasu, 2015-11-17:
I just checked that the minimum value permitted for illum_power_percentage
is 48 and maximum 100. That's the reason for failure.

In VoxelCLI, if you do

cap illum_power_percentage

when you connect a TintinCDK, you should get this information.

larrylisky, 2015-11-17:
That worked!

Thanks,

-Larry

larrylisky, 2015-11-17:
Requirement to access the camera specific header is no longer there for now.

closed 2015-11-17 by larrylisky