3dtof/voxelsdk #44: Incomplete data in XYZIPointCloudFrame

christophk86, 2015-12-01:
Hi everybody,
I get data from the camera via the following callback:

`<depthCamera->registerCallback(DepthCamera::FRAME_XYZI_POINT_CLOUD_FRAME, [&](DepthCamera &dc, const Frame &frame, DepthCamera::FrameType c)
{
    const XYZIPointCloudFrame *d_ = dynamic_cast<const XYZIPointCloudFrame *>(&frame);
...
}>'

The data in the pointer d is "inf". Only the d->points.at(i).i is valid.
Is there a reason why d->points.at(i).x, d->points.at(i).y and d->points.at(i).z are "inf"?
What do I have do to for getting the x,y and z data?

hlprasu, 2015-12-01:
> const XYZIPointCloudFrame *d_ = dynamic_cast(&frame);

This could happen if lens calibration used is incorrect.

Could you please provide the following:
1. Which depth camera are you using?
2. Lens calibration parameters used. Please provide the contents of camera
profile (.conf).

christophk86, 2015-12-01:
Thank you for your quick response.
I wrote an own application depending on this framework.

1) I use the Haddock depth camera from TI.
2) I think the Haddock config files will be used? Or does it depends from the name of my program?
[HaddockCDKCamera.conf.txt](https://github.com/3dtof/voxelsdk/files/48905/HaddockCDKCamera.conf.txt)
[HaddockCDKCameraNormal.conf.txt](https://github.com/3dtof/voxelsdk/files/48906/HaddockCDKCameraNormal.conf.txt)

Which parameters I have to modifiy for camera calibration?

hlprasu, 2015-12-02:
1. Could you please confirm whether ".txt" is actually present for the
configuration files? Configuration files are supposed to have only ".conf"
extension.

The content of those two configuration files are correct.

2. Are you using v0.6.0 version of SDK or older one?

christophk86, 2015-12-02:
Regarding 1) I add the txt-extension to enable the  file upload.
Regarding 2) I use version  0.5.7 of the SDK

3) Which parameters do I have to modifiy for camera calibration?

hlprasu, 2015-12-02:
The parameters are looking fine.

Could you please use the voxel viewer once and see whether you get point
cloud shown in it?

*@Anand:* any thoughts?

christophk86, 2015-12-02:
The VoxelCLI. displays no pointcloud, only a coordinate cross.
The consol output of the VoxelCLI is:
INFO: CameraSystem: Successfully loaded factory from library C:\Program Files\VoxelCommon\lib\haddockcdk.dll
INFO: CameraSystem: Successfully loaded factory from library C:\Program Files\VoxelCommon\lib\rts5825camera.dll
WARNING: CameraSystem: Ignoring Voxel library C:\Program Files\VoxelCommon\lib\ti3dtof.dll with ABI version = 27. Expected ABI version = 28
INFO: CameraSystem: Successfully loaded factory from library C:\Program Files\VoxelCommon\lib\tintincdk.dll
WARNING: CameraSystem: Ignoring Voxel library C:\Program Files\VoxelCommon\lib\voxel.dll with ABI version = 27. Expected ABI version = 28
INFO: CameraSystem: Successfully loaded factory from library C:\Program Files\VoxelCommon\lib\voxel14.dll
INFO: CameraSystem: Successfully loaded factory from library C:\Program Files\VoxelCommon\lib\voxeld.dll
ERROR: MainConfigurationFile: Default profile not valid in HaddockCDKCamera.conf
INFO: ParameterDMLParser: Found register map with name 'OPT9220A' (vPG1P0)

ERROR: DepthCamera: Could not set the camera profile to '128'
INFO: CameraSystem: Successfully refreshed parameters for HaddockCDKCamera(0::0451:9100::CK91001001235200122T).
INFO: ToFCameraBase: Starting with 320x240@30fps
_dealiasedPhaseMask = 0, _dealiasedPhaseMaskInPhaseOffset = 0
voxel:HaddockCDKCamera(0::0451:9100::CK91001001235200122T)> start
INFO: DepthCamera: Streaming stopped.
WARNING: DepthCamera: HaddockCDKCamera(0::0451:9100::CK91001001235200122T) already has a callback for this type = 3. Overwriting it now.
WARNING: DepthCamera: HaddockCDKCamera(0::0451:9100::CK91001001235200122T) already has a callback for this type = 2. Overwriting it now.
WARNING: DepthCamera: HaddockCDKCamera(0::0451:9100::CK91001001235200122T) already has a callback for this type = 1. Overwriting it now.
WARNING: DepthCamera: HaddockCDKCamera(0::0451:9100::CK91001001235200122T) already has a callback for this type = 0. Overwriting it now.
INFO: ToFCameraBase: Starting with 320x240@30fps
voxel:HaddockCDKCamera(0::0451:9100::CK91001001235200122T)>

hlprasu, 2015-12-03:
Do you have any other version of Voxel SDK installed?

Are there any files in your home directory ~/.Voxel/conf related to Haddock
CDK? If so, remove them and try.

NOTE: '~' corresponds to your home directory.

christophk86, 2015-12-03:
For testing with VoxelCLI I install Voxel SDK 0.5.7. It is the only framework on the windows PC. (My own program runs in a virtual mashine).
I use the TI 3D-TOF-CDK-EVM. Is it possible that it is the Haddack CDK Camera instead of the supposed Tintin depth camera?
The HaddockCDKCamera.conf exists in the C:\Program Files\VoxelCommon\conf directory.

gadiyar, 2015-12-04:
From the USB VID:PID (0x0451:0x9100), it looks like your camera is the Haddock CDK and not the Tintin CDK. (The TI 3D-TOF-CDK-EVM is the Haddock CDK Camera.)

The error messages above seem to indicate that you have multiple SDK versions installed and there might be some conflicts due to that. Could you confirm that you do not have more than one SDK version installed? (Maybe one installed from a pre-built package, and one that you built yourself?)

The SDK (v0.5.7 and earlier) looks for it's configuration files in the following locations:
1) Directory pointed to by the VOXEL_CONF_PATH environment variable
2) C:\Program Files\VoxelCommon\conf
3) %HOMEDRIVE%%HOMEPATH%\.Voxel\conf (usually maps to C:\Users\<userid>\.Voxel\conf)

Could you please check in these locations for any Haddock*.conf files and see if there are conflicting ones? The HaddockCDKCamera.conf file lists the filenames of all the other conf files that are applicable for a camera. These files have a profile id listed at the top of the file.

Please check for profile files with profile ID 128 and post it here as it seems to be the one with a conflict.

christophk86, 2015-12-07:
Thank you very much for your help.
I reinstall the Voxel SDK but the error is the same.
Then I copy the HaddockCDKCamera.conf in the bin-directory, then it works.
So next step was to see, if the voxel sdk installer sets the system variable VOXEL_CONF_PATH. The VOXEL_CONF_PATH was not set. So I add it manually - everything is fine.

closed 2015-12-07 by christophk86