3dtof/voxelsdk #46: How to use voxelsdk without cmake?

JoeyZhu, 2015-12-09:
Can I use voxelsdk with makefile only? Because I am wrapping voxelsdk into another sdk, which didn't use cmake system. Thank you !

hlprasu, 2015-12-09:
Yes, you can, though you'll have to take care of some things. I would
suggest you to build Voxel SDK as is using cmake and then copy necessary
headers and libraries as explained below.

1. Build SDK and install it into a folder /path/to/folder as follows in
build folder inside Voxel SDK.
---- For Linux ---
cmake -DCMAKE_INSTALL_PREFIX=/path/to/folder ..
sudo make install
---- For Windows ---
cmake -DCMAKE_INSTALL_PREFIX=/path/to/folder ..
cmake --target INSTALL --config Release --build .
-----

2. Copy /path/to/folder to your new SDK, and then in your custom make
system, set the following
-- Add include path -- /path/to/folder/include/voxel-x.y.z/
-- Link main program to -- /path/to/folder/lib/libvoxel.so (in Linux),
/path/to/folder/lib/voxel.dll (in Windows)

3. To run your program, please set environment variables as explained in
https://github.com/3dtof/voxelsdk/wiki/Build-from-source-on-Linux#running-from-source
for Linux and
https://github.com/3dtof/voxelsdk/wiki/Build-from-source-on-Windows#running-sdk-programs
for Windows.

Hope this helps.

JoeyZhu, 2015-12-11:
hlproasu, Thank you for your detail instruction, I successed to compile my DepthCapture_demo.cpp with g++ DepthCapture_demo.cpp -I /usr/include/voxel/ -L/usr/lib/voxel/ -lvoxel -std=c++11

closed 2015-12-11 by JoeyZhu