3dtof/voxelsdk #48: Windows build: keep separate output directories for build outputs

gadiyar, 2015-12-11:
This would allow switching betweeen release and non-release builds without overwriting all files each time. We can do this by generating the output file structure under a build-name directory instead of the top-level build directory.

(The current behavior is a regression from 0.5.7 due to the restructuring of the output directories).

hlprasu, 2015-12-11:
I've kept it that purposefully so that VOXEL_SDK_PATH can be just pointed
to the build directory and used irrespective of what configuration you
choose.

If you want to maintain simultaneously release and debug builds, I would
suggest to create separate build directories itself for them.

gadiyar, 2015-12-11:
VS configurations anyway allow you to use a different environment for each build.

Separate copies of the build directories requires two copies of the project to be opened in VS and requires one to make the same source modifications in two places. Normal workflow is to simply switch the build environment.