3dtof/voxelsdk #51: Failed to build from source v0.6.0 on Linux

JAHAHA, 2015-12-14:
Failed to build from source v0.6.0 on Linux. 
The following is the error message.
```````````````````````````````````````````````````````````````````````````````````````````
CMake Error at config/common.cmake:4 (export):
  export called with target "voxelpcl" which requires target "voxel" that is
  not in the export list.

  If the required target is not easy to reference in this call, consider
  using the APPEND option with multiple separate calls.
Call Stack (most recent call first):
  VoxelPCL/CMakeLists.txt:66 (create_cmake_config)

CMake Error at config/common.cmake:4 (export):
  export called with target "ti3dtof" which requires target "voxel" that is
  not in the export list.

  If the required target is not easy to reference in this call, consider
  using the APPEND option with multiple separate calls.
Call Stack (most recent call first):
  TI3DToF/CMakeLists.txt:148 (create_cmake_config)

hlprasu, 2015-12-14:
You'll need to update your cmake version to 2.8.13 or higher. Instructions
are available on wiki link -
https://github.com/3dtof/voxelsdk/wiki/Build-from-source-on-Linux

~ Prasad Bhat

JAHAHA, 2015-12-14:
Thanks a lot. It works due to use cmake 3.2. The previous cmake works fine with source v0.5.7, so I didn't try to upgrade the cmake to v3.2.

hlprasu, 2015-12-14:
Ok. We are making a minor change to enforce the CMake version requirement
in CMakeLists.txt.

~ Prasad Bhat

closed 2015-12-14 by JAHAHA