3dtof/voxelsdk #52: Get/Set configuration parameter -> memry access error

christophk86, 2015-12-27:
Hi everybody,

I want to read and write configuration parameters without using the configuration files.
I use the "get" and "getParam" functions but both functions cause to a memory access error.

My debug results are:
The function "_parameters.find(name)" raises the memory access error.

ParameterPtr param = depthCamera->getParam("unambiguous_range");
memory access error
const ParameterPtr DepthCamera::getParam(const String &name) const
{
  auto p = _parameters.find(name); // function raises the memory access error
...
}

What is wrong with this function or its usage?
Could you please help me to solve this quite strange error.

Thank you in advance.
Chris

hlprasu, 2015-12-27:
Hi Chris,

I do not see anything wrong with this in particular. It seems that error
must be due to something else.

Could you please a short program which reproduces this error?

Regards,
Prasad

gadiyar, 2017-01-04:
Closing - please reopen if this is still an issue.

closed 2017-01-04 by gadiyar