3dtof/voxelsdk #53: SimpleVoxelViewer doesn't work on linux if changed quirks

JAHAHA, 2016-01-12:
SimpleVoxelViewer show "ERROR: No data available. Waited for 2000 ms" and cannot find any data on PCL viewer, if we modified the parameter of uvcvideo by command "sudo modprobe  uvcvideo quirks=640". The following is the reproduce steps.
1. sudo rmmod uvcvideo
2. sudo modprobe uvcvideo quirks=640
3. plug-in device
4. run SimpleVoxelViewer
The below is output message.
---------------------------------------------------------------------
./SimpleVoxelViewer 
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libvoxeld.so.0.6.0
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libtintincdk.so.0.6.0
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/librts5825camera.so.0.6.0
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libhaddockcdk.so.0.6.0
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libvoxel14.so.0.6.0
INFO: UVC: Connecting to UVC device '/dev/video1'
ERROR: UVCXU: 0::0bda:5825::200901010001 UVCIOC_CTRL_QUERY failed.
WARNING: VoxelXUProgrammer: Could not get XU version.
INFO: UVC: Connecting to UVC device '/dev/video1'
INFO: UVC: Connecting to UVC device '/dev/video1'
INFO: UVCStreamer: 0::0bda:5825::200901010001 supports streaming modes
INFO: UVCStreamer: 0::0bda:5825::200901010001 supports mmap
INFO: ParameterDMLParser: Found register map with name 'OPT9220A' (vPG1P0)

ERROR: UVCXU: 0::0bda:5825::200901010001 UVCIOC_CTRL_QUERY failed.
ERROR: VoxelXUProgrammer: Could not set read register address 0x5c78
ERROR: VoxelProgrammerBase: Could not read register for address 0x5c78
ERROR: DepthCamera:Could not get value for parameter RTS5825Camera(0::0bda:5825::200901010001).scratch2
INFO: DepthCamera: Setting parameter 'unambiguous_range'
INFO: DepthCamera: Setting parameter 'sub_frame_cnt_max'
INFO: DepthCamera: Setting parameter 'quad_cnt_max'
INFO: DepthCamera: Setting parameter 'intg_time'
INFO: DepthCamera: Setting parameter 'illum_volt'
INFO: CameraSystem: Successfully refreshed parameters for RTS5825Camera(0::0bda:5825::200901010001).
INFO: ToFCameraBase: Starting with 320x240@25fps
_dealiasedPhaseMask = 0, _dealiasedPhaseMaskInPhaseOffset = 0
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms
ERROR: No data available. Waited for 2000 ms

hlprasu, 2016-01-12:
Could you please explain more about the quirk "modprobe uvcvideo
quirks=640" -- Why you are using it and what does it do?

JAHAHA, 2016-01-14:
There are two cameras on our device. One is RBG, another is TOF. Running these two cameras in Ubuntu14.04 results in the the following error:
libv4l2: error turning on stream: No space left on device
VIDIOC_STREAMON: No space left on device

"quirks=128" forces uvcvideo driver to  calculate their bandwidth usage instead of requesting their maximum.

hlprasu, 2016-01-14:
Underlying primary problem as I understand is that there two devices which
are connected to one USB host controller and hence they are being
arbitrated for bus access. Both these devices in this case are pumping
large data periodically causing this issue.

Three options:
1. Does quirks=128 itself work out for you, rather than quirks=640?
2. See whether your RGB camera supports compressed format such as MPEG or
MJPEG. If so, use it for that. See
http://superuser.com/questions/431759/using-multiple-usb-webcams-in-linux
3. See whether you can plug the two cameras to different USB host
controllers. Typically one PC or laptop has at least two USB host
controllers whose physical ports are at different locations. An adjacent
pair of USB ports are typically going to one USB host controller. Avoid
that and try out different ports.

JAHAHA, 2016-01-15:
Actually our device includes two cameras that are connected to one USB hub that connect PC via USB cable. So I cannot connect our device to different USB host controller. 

My question is why the voxel SDK cannot work if quirks=128? Even if I doesn't run RGB camera.

gadiyar, 2016-01-15:
Could you please confirm that the camera works if you do not pass the quirks=128 parameter?
(We want to ensure the camera is functional by itself).

Also, please let us know your platform (PC or ARM?), Linux distribution and kernel version (output from `uname -a` command will do).

JAHAHA, 2016-01-15:
Yes, the camera works fine if don't pass the quirks=128 parameter. 
My platform is Ubuntu 14.04(Intel PC) and kernel version is 3.13.0-32-generic.

hlprasu, 2016-01-15:
Is your problem that the RGB camera is not working without the quirks?

JAHAHA, 2016-01-15:
I depends on which camera starts to work first. If our application start to run TOF camera, then RGB doesn't work due to "No space left on device". Otherwise TOF camera doesn't work.

gadiyar, 2016-01-15:
Okay, we can reproduce the issue with quirks=128, but I am not sure this is the right quirk to use.

Could you please let us know what mode you use the RGB camera in (MJPEG or YUY2), and at what resolution?

gadiyar, 2016-01-15:
From the code, it looks like setting this quirk causes the dwMaxPayloadTransferSize to be recalculated and programmed to a different value than the default.
http://lxr.free-electrons.com/source/drivers/media/usb/uvc/uvc_video.c#L162

This might not be supported correctly by the camera firmware. Please contact your camera chipset vendor (Realtek) to investigate this - this is not related to the VoxelSDK.

JAHAHA, 2016-01-15:
You are right about uvcvideo driver. 
I used another application "guvcview" to double check this issue.  I found that this application is able to receive the data from TOF camera, but VoxelSDK show "ERROR: No data available. Waited for 2000 ms" at the same test scenario. So if I report it to USB chip vendor, I think they should use the general application to verify this issue as well.   
Do you have idea why the general webcam viewer can receive the data with quirks=128?

gadiyar, 2016-01-15:
Okay, I can reproduce this too. guvcview can stream data while VoxelSDK cannot.

JAHAHA, 2016-01-15:
What USB chip-set is on your board? Is it Realtek, too?

gadiyar, 2016-01-15:
I have the same board with the Realtek chipset (USB VID:PID 0bda:5825 for TOF).

JAHAHA, 2016-01-18:
Are you still investigating this issue and then provide the solution?

gadiyar, 2016-01-18:
We are investigating this but it might take some time. For now, we don't have any workaround other than to not use that quirk.

hlprasu, 2016-03-02:
Some observations: 
1. I tried to use Voxel-A (Voxel14) with quirks=128 and it works fine with SDK.
2. I've checked and seen that there is no different between information reported by v4l2-ctl for Voxel-A and yours.
3. guvcview is unable to get stream from yours for a few attempts, after which it gets stream successfully.  This is not case with guvcview streaming from Voxel-A. Need to see why.

It needs some more analysis.