3dtof/voxelsdk #54: frames dropped when registering pointcloudXYZI callback

cnanda-zbr, 2016-01-15:
Hi,
I am registering a callback to get pointcloudXYZI data from the Tin-Tin CDK and get the following warning. "Dropping frame due to slow forward pipeline" . I am not doing anything in the callback yet. I tried to change the default frame buffer size from 2 ( why such a low number ? ) to a higher number 64. 

It still does not solve the problem. 

Another question, you are using a list ( not non-blocking DS) , and performing a dynamic memory allocation  in USBBulkStreamer.cpp, could that be causing it to block  ? 

-Best Regards,

Chinmay

hlprasu, 2016-01-16:
Could you please share a minimalistic sample source code which demonstrates
this issue?

Also, please share details about your machine.

About the frame buffer architecture -
1. Typically the whole pipeline takes less than 5% of CPU on a 2.5GHz quad
core machine. So, frame buffer size 2 is just enough.
2. If you read through USBBulkStreamer.cpp, especially
USBBulkStreamer::USBBulkStreamerPrivate::putBuffer() and
USBBulkStreamer::USBBulkStreamerPrivate::getBuffer(), you'll see that
dynamic memory is allocated only in the beginning and then reused every
time.

Regards
Prasad

~ Prasad Bhat

cnanda-zbr, 2016-01-20:
Hi Prasad,

Sorry I got caught up with something else so could not post my reply. 
My hardware details: Intel NUC i7 16gb RAM , running Ubuntu 14.04. 

I am using CLIManager demo app as a starting point. I was using the following code to initialize the callback. 
Ptr<pcl::Grabber> _grabber; // This will link to our Voxel::PCLGrabber

 _grabber = Ptr<pcl::Grabber>(new Voxel::PCLGrabber(*_currentDepthCamera));
      boost::function<void (const pcl::PointCloud<pcl::PointXYZI> &)> f = boost::bind(&CLIManager::_cloudBufferCallback, this, _1);
     _grabber->registerCallback(f);
     _grabber->start();

In the callback.
void CLIManager::_cloudBufferCallback(const pcl::PointCloud<pcl::PointXYZI> &cloud)
{
}

causes the printout that I earlier mentioned. 

-Best Regards,
Chinmay

hlprasu, 2016-01-21:
> boost::function &)> f = boost::bind(&CLIManager::_cloudBufferCallback,

CLIManager already has a grabber connected to the same depth camera. So, it
might cause conflicts here.

I would suggest you to use CameraSystemTest.cpp in "Test" folder of Voxel
SDK source as starting point for trying out callbacks and capture. It has
all possible callback types illustrated but within comment blocks.

cnanda-zbr, 2016-01-21:
Hi, 
I am porting the code to ROS and found the issue was somewhere else. The code works fine the way it is. 
Many thanks!.

-Chinmay

closed 2016-01-21 by cnanda-zbr

zeeraq, 2016-02-15:
@cnanda-zbr 
Hi Chinmay,

Can you please tell me how you solved your problem? I'm also having the same problem as you.

-Zeerak

cnanda-zbr, 2016-02-15:
Hi Zeerak,
I installed the SDK on an intel i7 and ported  the code over to ROS. I suspect that the hardware was falling short of CPU resources when running it at 30fps and with the hardware filters. We are running the CDK at 5fps , this does not cause the issue any more. One thing that I did was to move the code in the context of ROS instead of creating an object and then calling the constructor from the library. 
This was something specific to ROS not sure what you are trying to do or what problem you may have, 

-Chinmay