3dtof/voxelsdk #55: setFrameRate() makes depth data abnormal

JoeyZhu, 2016-01-26:
I set framerate before depthCamera->start() with

    FrameRate r;
    r.numerator = 15.0f * 10000;
    r.denominator = 10000;
    uint g = gcd(r.numerator, r.denominator);
    r.numerator = r.numerator/g;
    r.denominator = r.denominator/g;

but depthdata is abnormal, see the link below.

when I commented out code above, depth data is ok.

good depth data:
http://pan.baidu.com/s/1bmGoeM

bad depth data:
http://pan.baidu.com/s/1o7w1h8m

Thank you !

hlprasu, 2016-01-26:
Could you please provide a minimalistic runnable code which demonstrates
this problem?

Are you using the setFrameRate() API or setting parameters directly?

Which depth camera are you using? TintinCDK?

Thanks
Prasad

~ Prasad Bhat

JoeyZhu, 2016-01-26:
After referring to 9221 datasheet(SBAS703A), I found INTG_DUTY_CYCLE should be updated according to the frame rate settings.
So after calling setFrameRate, I called set() API to set INTG_DUTY_CYCLE to a half of previous value due to I set frame rate from 30 to 15. And the depthdata is ok now.

Yes, I'm using TI's TintinCDK. And I use setFrameRate() API to set frame rate.

I think setFrameRate() API should contain this kind of low level parameters change, right ? Thank you!

hlprasu, 2016-01-26:
I just checked to confirm whether setFrameRate() refreshes integer duty
cycle or not. It actually does it via integration % time parameter.

See https://github.com/3dtof/voxelsdk/blob/master/TI3DToF/ToFCamera.cpp#L175

@Anand: Any thoughts?