3dtof/voxelsdk #61: how many LUT will be needed for a nonlinear test?

kevinliu-ez, 2016-02-01:
take an example, if 48MHz, if 10 different positions that we like to measure the phase offset, (if testing chart vs camera postion), then, should we need to have 10 x (240x320) LUT table? where is your Voxel sdk call those tables?---tnks for your answers!

larrylisky, 2016-09-04:
Nonlinearity compensation is done by OPT9221 online.  You will need to program the 16-entry lookup table value. The LUT is phase-to-phase (using nonlinear phase to lookup ideal phase).

kevinliu-ez, 2016-09-20:
Hello, Larry, could you help let us know whether to get related calibration
kit???
tnks a lot!!!

Kevin Liu，Ph.D
GM, eZimage Technology Co. ltd
易積生活科技
0919377901

larrylisky, 2016-09-20:
Nonlinear calibration is part of VoxelViewer.  We don't currently have a kit to offer. Typically one can use a linear positioning stage to perform such calibration. 

-Larry

gadiyar, 2017-06-20:
Nothing to do here.

closed 2017-06-20 by gadiyar