3dtof/voxelsdk #62: Default filter bank to boost point-cloud quality.

byfron, 2016-02-01:
The windows original TI voxel viewer achieves a very nice point-cloud quality (outlier free) out-of-the-box probably due to a nice toolchain of filters that are set by default. It'd be nice to include an example/test where the point cloud is already pre-processed by such filter bank. I'm doing some tests right now with TintinCDK so I'll be happy to contribute if I succeed.

hlprasu, 2016-02-02:
By default, there are no filters applied. However, a default camera profile
is applied which sets various calibrations. The default camera profile is
applied whether you use SDK or viewer.

SDK has API to select camera profile while viewer has a menu to select the
same.

~ Prasad Bhat

byfron, 2016-02-02:
Hi Prasad, thanks for your answer. So after playing around I realized the is the amplitude threshold the parameter that was removing the outliers in the windows gui, so no filtering indeed.

closed 2016-04-17 by hlprasu