3dtof/voxelsdk #63: Applying filters on recorded data

byfron, 2016-02-01:
It seems that applying filters to pre-recorded VXL raw sequences is not straightforward. The FrameStreamReader class does not implement this functionality as far as I know.

hlprasu, 2016-02-02:
Yes, true. It is not supported.

~ Prasad Bhat

byfron, 2016-02-02:
It'd be really useful to be able to process raw vxl data as if there was a "virtual" camera so that all the regular camera parameters as well as filters can be applied.

hlprasu, 2016-02-03:
Noted. Filters are possible to implement though not in the plan yet.

Parameters will be practically not possible as they are meant to influence
the initial raw data itself and not subsequent processing.

byfron, 2016-02-03:
Although I assume certain parameters like illumination power thresholds could be applied in pre-recorded raw data right?. At least the windows Voxel viewer gui can do this. 

Thanks anyway and looking forward to those improvements.

hlprasu, 2016-02-03:
In order to apply the parameters such as illumination power, we will need
to fully simulate the sensor board itself which is not the target of Voxel
SDK. The VXL format is a playback format and not really a simulation format.

~ Prasad Bhat

closed 2016-09-05 by larrylisky