3dtof/voxelsdk #64: programming frequency calibration parameter

cnanda-zbr, 2016-02-02:
Hi,
1.After performing frequency calibration I get a floating point number. What parameter is used to apply that to pointcloud data ? I assumed it is "frequency_scale". But it only accepts uint values. So what is the right way to apply this calibration parameter ? Does it get applied in the pointcloud data if I register for a callback ?  Or do I have to do subscribe for a depth callback and then do the correction myself ?

2.       The maximum value of intg_duty_cycle that I can set programmatically is 20. Anything above that , gives me an error saying that “unable to set the parameter” .  The windows GUI allows you to go all the way upto 31 for 30 fps.

3. Is there a source code for VoxelViewer source on Linux ? I am developing an application and that can help answer some of my questions. 

-Best Regards,
Chinmay

gadiyar, 2016-02-03:
`frequency_scale` is a test-pattern related parameter.
`freq_corr_at` is the parameter that needs to be set for adjusting the modulation_frequency scaling factor. I will upload a snippet from a .conf file shortly with this setting.

The maximum value of intg_duty_cycle for the TintinCDK was limited to 20 in the VoxelSDK, and the same restriction applies for VoxelViewer as well (because it uses the VoxelSDK directly). Note that there are two parameters - one called `intg_duty_cycle` which is an OPT9221 register representing the integration duty cycle as a fraction (out of 64) of the frame time and one called `intg_time` which is a convenience parameter representing the integration duty cycle as a percentage. The VoxelViewer GUI shows the integration_time in the front panel and this parameter has the upper limit of 31. Setting either of these parameters sets the other one automatically.

The VoxelViewer source code is not available at this time, but I'll see if we can share some snippets from it separately. We plan to break out the calibration wizard portion separately as that is the most useful part.

hlprasu, 2016-02-03:
Snippets are being adding in this link -
https://github.com/3dtof/voxelsdk/wiki/Voxel-SDK-Calibration-Guide

You'll have it fully available in next 2-3 days.

cnanda-zbr, 2016-02-03:
Hi Anand/Prasad,
Thanks for your comments. 
So can I set "freq_corr_at" like other parameters ?
I do the following for setting other parameters.
 ParameterPtr param = _currentDepthCamera->getParam(paramString);

    BoolParameter *boolParam = dynamic_cast<BoolParameter *>(param.get());
    IntegerParameter *intParam = dynamic_cast<IntegerParameter *>(param.get());
    UnsignedIntegerParameter *uintParam = dynamic_cast<UnsignedIntegerParameter *>(param.get());
    FloatParameter *floatParam = dynamic_cast<FloatParameter *>(param.get());
    EnumParameter *enumParam = dynamic_cast<EnumParameter *>(param.get());

this code snippet is from the CLIManager. 

I get unknown parameter error when I look for  freq_corr_at.

-Chinmay

cnanda-zbr, 2016-02-12:
Hi Prasad,

Are you still in the process of updating the URLs?  

-Best Regards,
Chinmay