3dtof/voxelsdk #65: Set camera parameters via file -> parameter not found

christophk86, 2016-02-28:
Hi,
when I set the parameters e.g. "intg_time" via the HaddockCDKCameraNormal.conf file the following error comes up while starting my program.
_ERROR: DepthCamera:Could not get value for parameter HaddockCDKCamera(0::0451:9100::CK91001001235200122T).intg_time
ERROR: DepthCamera: Failed to update value for parameter 'unambiguous_range'
ERROR: DepthCamera: Failed to update value for parameter 'intg_time'
ERROR: DepthCamera:Could not get value for parameter HaddockCDKCamera(0::0451:9100::CK91001001235200122T).intg_time
ERROR: DepthCamera: Failed to update value for parameter 'unambiguous_range'
ERROR: DepthCamera: Failed to update value for parameter 'intg_time'
ERROR: DepthCamera:Could not get value for parameter HaddockCDKCamera(0::0451:9100::CK91001001235200122T).intg_time
ERROR: DepthCamera: Failed to update value for parameter 'unambiguous_range'
ERROR: DepthCamera: Failed to update value for parameter 'intg_time'
ERROR: CameraSystem: Could not refresh parameters for HaddockCDKCamera(0::0451:9100::CK91001001235200122T).
INFO: Successfully loaded depth camera for device 0::0451:9100::CK91001001235200122T
INFO: ToFCameraBase: Starting with 320x240@20fps_

If I set the same parameter with the VoxelCLI via the command set intg_time = 30 the parameter could be set correctly.

Could anybody please tell me the reason?
Thanks in advance.

hlprasu, 2016-02-28:
Could you share your modified conf file?

Also, please clarify whether streaming happens in the case where you set
parameter via conf file.

Thanks

christophk86, 2016-02-28:
My two modified conf files:
[HaddockCDKCameraNormal_Modified.zip](https://github.com/3dtof/voxelsdk/files/149993/HaddockCDKCameraNormal_Modified.zip)

The streaming starts with the modified conf files, but the error messages above.

hlprasu, 2016-02-28:
I think the problem is that you are setting intg_time to 100 which
integration duty cycle is 100%. That might be the cause of this issue. Can
you try setting for instance 30 in configuration and see whether this
problem recurs?

christophk86, 2016-02-28:
I restart the eval kit.
I set the integration duty cycle to 30. -> No error
I set the integration duty cycle back to 100 -> No error.

I check the illumination power by "get illum_power_percentage" in the VoxelCLI.
In the conf file it set to 75, but the return value from "get illum_power_percentage" was 20%.

It seems that the values of the conf file are not set.

hlprasu, 2016-02-28:
Could you please switch off/on the board (NOT RESET) and try again?

The current profile ID is stored in temporary register. If applied profile
ID is same as current profile ID then parameters are not freshly applied
causing this confusion.

However, the actual problem is still than after last switch on, profile
with intg_time = 100 was tried which failed, but current profile ID got
saved. Subsequent changes in profile defining_params did not get applied by
design.

Thanks.

christophk86, 2016-02-28:
Ok, switch off/on solve the problem.
Thank you for the solution and explanation.

closed 2016-02-28 by christophk86