3dtof/voxelsdk #67: How to start voxel sdk programming with vs2013?

iammeizu, 2016-03-30:
I am a newcomer to voxel sdk, and I have already installed voxel sdk6.0 in my pc. I have run VoxelCLI.exe successfully.
What should I do next ? 
How to build a test project?
How to configure my vs 2013?
Do I need a TI‘s OPT8241-CDK-EVM ?

Thank you in advance !

hlprasu, 2016-03-30:
Please have a look at
https://github.com/3dtof/voxelsdk/wiki/Voxel-SDK-User-Guide

iammeizu, 2016-03-30:
I am using 64-bit win10, so I installed Voxelsdk not build from source on windows.
Dose this mean I must install voxelsdk again(from source)? 
Thank you again!

hlprasu, 2016-03-30:
No. Not necessary. However, please install all the prerequisites mentioned
in  https://github.com/3dtof/voxelsdk/wiki/Build-from-source-on-Windows in
order to compile the example given in
https://github.com/3dtof/voxelsdk/wiki/Voxel-SDK-User-Guide

closed 2016-04-05 by iammeizu