3dtof/voxelsdk #68: Run 'CameraSystemReadStreamTest', encounter an error

AlexCheun, 2016-03-31:
Hi all,
 After successfully compile the source code from github,  I tried to run the sample 'CameraSystemReadStreamTest' but encounter the error   "ERROR,Require argurement missing ".

Then, I tried to run the sample "SimpleVoxelViewer"， and got assert error, pointed to ' if(!ToFTintinCamera::_init())'  in TintinCDKCamera.cpp,  I stepped into the codes to check. I found the assert was thrown at the line  "i:second->refresh()" inside the following function.
   
 bool DepthCamera::refreshParams()
{
  Lock<Mutex> _(_accessMutex);
  bool ret = true;
  for(auto &i: _parameters)
  {
    if(!i.second->refresh())
    {
      logger(LOG_ERROR) << "DepthCamera: Failed to update value for parameter '" << i.first << "'" << std::endl;
      ret = false;
    }
  }
  return ret;
}

please kindly help me out for these. Thanks a bunch.

hlprasu, 2016-03-31:
> Lock _(_accessMutex);

Regarding the first one, you just need to pass the correct arguments. In
this case, it expects '-f' argument for reading a file.

For second one, could you please provide the complete error?

Assuming you are using Windows, have you set the necessary environment
variables as explained in
https://github.com/3dtof/voxelsdk/wiki/Build-from-source-on-Windows#running-sdk-programs
?

Regards
Prasad

AlexCheun, 2016-03-31:
Thanks for replying soon.  Regarding the first one,I pass the argument  "aaa.txt" , but return 
"ERROR: FrameStreamReader: Failed to read frame headers.
ERROR: File could not be opened for reading"

AlexCheun, 2016-03-31:
for the second one,  I do the setting environment according to https://github.com/3dtof/voxelsdk/wiki/Build-from-source-on-Windows#running-sdk-programs. In addition, I add environment  variables ”VOXEL_CONF_PATH “, "VOXEL_FW_PATH" ; "VOXEL_LIB_PATH"  and add the path accordingly.  the program throw out  access failure. 

Another problem occurs now,  I cannot power up the board after connect the 5DC power and usb cable into the borad. The read led blinking but cannot see green LED and blue LED lighted....

hlprasu, 2016-03-31:
For the first one, you are expected to pass a VXL file.

For the second one, could you please give the exact error? What does it
show in terminal and if any dialogs what do they say exactly?

AlexCheun, 2016-04-01:
Hi Bhat,
I cannot power up the board now. I run the Voxel  Viewer-0,7,0 but cannot scan and find out CDK board. How can I solve this before showing the second debug info to you. Thanks a lot.

Alex

AlexCheun, 2016-04-02:
after change the 5DC1A power to 5DC3A as recommended, all leds work fine. The second SDK sample output in console are 
WARNING: CameraSystem: Failed to load or register a depth camera factory from library E:\TI+solution\TI solution\voxelsrc_build02\lib\ti3dtof.dll. Ignoring this library.
WARNING: CameraSystem: Failed to load or register a depth camera factory from library E:\TI+solution\TI solution\voxelsrc_build02\lib\voxel.dll. Ignoring this library.
WARNING: CameraSystem: Ignoring Voxel library E:\TI+solution\TI solution\voxelsrc_build02\lib\voxelpcl.dll with ABI version = 0. Expected ABI version = 29
INFO: CameraSystem: Successfully loaded factory from library E:\TI+solution\TI solution\voxelsrc_build02\lib\voxel\calculuscdk.dll
INFO: CameraSystem: Successfully loaded factory from library E:\TI+solution\TI solution\voxelsrc_build02\lib\voxel\haddockcdk.dll
INFO: CameraSystem: Successfully loaded factory from library E:\TI+solution\TI solution\voxelsrc_build02\lib\voxel\rts5825camera.dll
INFO: CameraSystem: Successfully loaded factory from library E:\TI+solution\TI solution\voxelsrc_build02\lib\voxel\tintincdk.dll
INFO: CameraSystem: Successfully loaded factory from library E:\TI+solution\TI solution\voxelsrc_build02\lib\voxel\voxel14.dll
INFO: CameraSystem: Successfully loaded factory from library E:\TI+solution\TI solution\voxelsrc_build02\lib\voxel\voxeld.dll
INFO: TintinCDKCamera: Board revision = 2.0
INFO: ParameterDMLParser: Found register map with name 'OPT9221' (vPG1P0)

INFO: TintinCDKCamera: Received 47033 bytes from EEPROM in 2.11733 s
INFO: Data2DCodec: Original number of 8-bit offsets = 4383
INFO: Data2DCodec: Number of invalid pixels in original data = 0
INFO: Data2DCodec: Current number of 8-bit offsets = 4383

after this,  the program throws out 0x01251B68 unprocess execption at (在 SimpleVoxelViewer.exe ):  0xC0000005: position  0x01251B68  access  .vialation

hlprasu, 2016-04-02:
Have you built it with release mode or debug mode? Could you please
confirm?

AlexCheun, 2016-04-02:
built it in debug mode,  I double check the "Running SDK programs" in tuition,  I find I missed a part which is        

"Due to a bug, in Visual Studio 2013, after you have set the required environment variables, you need to close the Visual studio and edit the ".user" file corresponding to the target, and replace all newlines within with &#xd;&#xa"

would it effect?

hlprasu, 2016-04-02:
> with &#xa"

You'll need to use PCL built in debug mode
(PCL-1.7.2-AllInOne-win64-Debug.exe) from
https://github.com/3dtof/pcl/releases/tag/1.7.2.

This is mentioned a bit indirectly in -
https://github.com/3dtof/voxelsdk/wiki/Installation%20on%20Windows#prerequisites

Regards,
Prasad

AlexCheun, 2016-04-02:
hi ,
    i re-built the source code with x64 debug version of PCL, and run 'simple voxel viewer ' sample again. Encounter errors like this.
    WARNING: CameraSystem: Failed to load or register a depth camera factory from library E:\voxelsdk\build\lib\ti3dtof.dll. Ignoring this library.
WARNING: CameraSystem: Failed to load or register a depth camera factory from library E:\voxelsdk\build\lib\voxel.dll. Ignoring this library.
WARNING: CameraSystem: Ignoring Voxel library E:\voxelsdk\build\lib\voxelpcl.dll with ABI version = 0. Expected ABI version = 29
ERROR: DepthCameraLibrary: Failed to load E:\voxelsdk\build\lib\voxel\calculuscdk.dll. 

WARNING: CameraSystem: Failed to load library E:\voxelsdk\build\lib\voxel\calculuscdk.dll. Ignoring it.
ERROR: DepthCameraLibrary: Failed to load E:\voxelsdk\build\lib\voxel\haddockcdk.dll. 

WARNING: CameraSystem: Failed to load library E:\voxelsdk\build\lib\voxel\haddockcdk.dll. Ignoring it.
ERROR: DepthCameraLibrary: Failed to load E:\voxelsdk\build\lib\voxel\rts5825camera.dll. 

WARNING: CameraSystem: Failed to load library E:\voxelsdk\build\lib\voxel\rts5825camera.dll. Ignoring it.
ERROR: DepthCameraLibrary: Failed to load E:\voxelsdk\build\lib\voxel\tintincdk.dll. 

WARNING: CameraSystem: Failed to load library E:\voxelsdk\build\lib\voxel\tintincdk.dll. Ignoring it.
ERROR: DepthCameraLibrary: Failed to load E:\voxelsdk\build\lib\voxel\voxel14.dll. 

WARNING: CameraSystem: Failed to load library E:\voxelsdk\build\lib\voxel\voxel14.dll. Ignoring it.
ERROR: DepthCameraLibrary: Failed to load E:\voxelsdk\build\lib\voxel\voxeld.dll.

WARNING: CameraSystem: Failed to load library E:\voxelsdk\build\lib\voxel\voxeld.dll. Ignoring it.
WARNING: CameraSystem: No depth camera library found or loaded.

But in the lib\voxel folder ,  there exist
 calculuscdk.dll, 
haddockcdk.dll,
rts5825camera.dll,
tintincdk.dll,
voxel14.dll,
voxeld.dll

And I tracked the code into 'DepthCameraLibrary.cpp' , in fuction ''bool DepthCameraLibrary::load()

if(!_libraryPrivate->handle) 
  {
    logger(LOG_ERROR) << "DepthCameraLibrary: Failed to load " << _libName << ". Error: " << dynamicLoadError() << std::endl;
    return false;
  }

The _libraryPrivate->handle is NULL ...

Thanks

gadiyar, 2016-04-02:
Please make sure you do not have any previous version of VoxelSDK binaries in your path. You can use 'DependencyWalker' to examine. Usually, such issues are caused by having a different version of ti3dtof.dll and/or voxel.dll in the path.

From: AlexCheun
Sent: Saturday, April 2, 2016 16:56
To: 3dtof/voxelsdk
Reply To: 3dtof/voxelsdk
Subject: Re: [3dtof/voxelsdk] Run 'CameraSystemReadStreamTest', encounter an error (#68)

hi ,
i re-built the source code with x64 debug version of PCL, and run 'simple voxel viewer ' sample again. Encounter errors like this.
WARNING: CameraSystem: Failed to load or register a depth camera factory from library E:\voxelsdk\build\lib\ti3dtof.dll. Ignoring this library.
WARNING: CameraSystem: Failed to load or register a depth camera factory from library E:\voxelsdk\build\lib\voxel.dll. Ignoring this library.
WARNING: CameraSystem: Ignoring Voxel library E:\voxelsdk\build\lib\voxelpcl.dll with ABI version = 0. Expected ABI version = 29
ERROR: DepthCameraLibrary: Failed to load E:\voxelsdk\build\lib\voxel\calculuscdk.dll.

WARNING: CameraSystem: Failed to load library E:\voxelsdk\build\lib\voxel\calculuscdk.dll. Ignoring it.
ERROR: DepthCameraLibrary: Failed to load E:\voxelsdk\build\lib\voxel\haddockcdk.dll.

WARNING: CameraSystem: Failed to load library E:\voxelsdk\build\lib\voxel\haddockcdk.dll. Ignoring it.
ERROR: DepthCameraLibrary: Failed to load E:\voxelsdk\build\lib\voxel\rts5825camera.dll.

WARNING: CameraSystem: Failed to load library E:\voxelsdk\build\lib\voxel\rts5825camera.dll. Ignoring it.
ERROR: DepthCameraLibrary: Failed to load E:\voxelsdk\build\lib\voxel\tintincdk.dll.

WARNING: CameraSystem: Failed to load library E:\voxelsdk\build\lib\voxel\tintincdk.dll. Ignoring it.
ERROR: DepthCameraLibrary: Failed to load E:\voxelsdk\build\lib\voxel\voxel14.dll.

WARNING: CameraSystem: Failed to load library E:\voxelsdk\build\lib\voxel\voxel14.dll. Ignoring it.
ERROR: DepthCameraLibrary: Failed to load E:\voxelsdk\build\lib\voxel\voxeld.dll.

WARNING: CameraSystem: Failed to load library E:\voxelsdk\build\lib\voxel\voxeld.dll. Ignoring it.
WARNING: CameraSystem: No depth camera library found or loaded.

But in the lib\voxel folder , there exist
calculuscdk.dll,
haddockcdk.dll,
rts5825camera.dll,
tintincdk.dll,
voxel14.dll,
voxeld.dll

And I tracked the code into 'DepthCameraLibrary.cpp' , in fuction ''bool DepthCameraLibrary::load()

if(!_libraryPrivate->handle)
{
logger(LOG_ERROR) << "DepthCameraLibrary: Failed to load " << _libName << ". Error: " << dynamicLoadError() << std::endl;
return false;
}

The _libraryPrivate->handle is NULL ...

Thanks

AlexCheun, 2016-04-04:
Hi gadiyar ,
    I had already unisntalled previous version of voxel sdk, there's only one in PATH.

gadiyar, 2016-04-04:
Yes, but please check using DependencyWalker to be sure

AlexCheun, 2016-04-04:
running the DependencyWalker , file -> open , select 'SimpleVoxelViewer.exe', a tree list shows up, 'KERNEL32.DLL'  
'ADVAPI32.DLL'
'VOXELPCL.DLL'
'VOXEL.DLL'
'MSVCP120D.DLL'
'MSVCR120D.DLL'

gadiyar, 2016-04-04:
Right - check the path to VOXEL.DLL and also TI3DTOF.DLL listed in the tree view. Make sure they match the version you built and not some other version.

AlexCheun, 2016-04-04:
There‘s no TI3DTOF.dll exsits in the tree view. Howerver,I rebuild all the projects and re-debug the simpleVoxelViewer project, information shows in console like this,
WARNING: CameraSystem: Failed to load or register a depth camera factory from library E:\voxelsdk\build\lib\ti3dtof.dll. Ignoring this library.
WARNING: CameraSystem: Failed to load or register a depth camera factory from library E:\voxelsdk\build\lib\voxel.dll. Ignoring this library.
WARNING: CameraSystem: Ignoring Voxel library E:\voxelsdk\build\lib\voxelpcl.dll with ABI version = 0. Expected ABI version = 29
INFO: CameraSystem: Successfully loaded factory from library E:\voxelsdk\build\lib\voxel\calculuscdk.dll
INFO: CameraSystem: Successfully loaded factory from library E:\voxelsdk\build\lib\voxel\haddockcdk.dll
INFO: CameraSystem: Successfully loaded factory from library E:\voxelsdk\build\lib\voxel\rts5825camera.dll
INFO: CameraSystem: Successfully loaded factory from library E:\voxelsdk\build\lib\voxel\tintincdk.dll
INFO: CameraSystem: Successfully loaded factory from library E:\voxelsdk\build\lib\voxel\voxel14.dll
INFO: CameraSystem: Successfully loaded factory from library E:\voxelsdk\build\lib\voxel\voxeld.dll
INFO: TintinCDKCamera: Board revision = 2.0
INFO: ParameterDMLParser: Found register map with name 'OPT9221' (vPG1P0)

INFO: TintinCDKCamera: Received 47033 bytes from EEPROM in 2.1247 s
INFO: Data2DCodec: Original number of 8-bit offsets = 4383
INFO: Data2DCodec: Number of invalid pixels in original data = 0
INFO: Data2DCodec: Current number of 8-bit offsets = 4383

The same as previous post as I asked Bhat.  He told me to re-compile the solution with PCL-64bit Debug Version and I diid it.

gadiyar, 2017-01-04:
Closing this item - please reopen if this is still an issue.

closed 2017-01-04 by gadiyar