3dtof/voxelsdk #69: No Makefile

FUT, 2016-04-03:
I was trying to build voxelsdk from source (https://github.com/3dtof/voxelsdk/wiki/Build-from-source-on-Linux#build-from-source), but was not able to run the tast command of the list: 

```
cd voxel-sdk
mkdir build
cd build
export CMAKE_PREFIX_PATH=$PWD:$CMAKE_PREFIX_PATH
cmake -DCMAKE_BUILD_TYPE=Release ..
make -j2       
```

When I am trying to run `make -j2` - I get an error:

```
make: *** No targets specified and no makefile found.  Stop.
```

I've inspected all the directories in the `build` folder but there is no `Makefile` also. Did the installation process changed?

gadiyar, 2016-04-03:
The makefiles should get generated when you run the cmake step just before calling make. Are you sure that step ran without errors?

From: Yury Kaliada
Sent: Sunday, April 3, 2016 19:48
To: 3dtof/voxelsdk
Reply To: 3dtof/voxelsdk
Subject: [3dtof/voxelsdk] No Makefile (#69)

I was trying to build voxelsdk from source (https://github.com/3dtof/voxelsdk/wiki/Build-from-source-on-Linux#build-from-source), but was not able to run the tast command of the list:

cd voxel-sdk
mkdir build
cd build
export CMAKE_PREFIX_PATH=$PWD:$CMAKE_PREFIX_PATH
cmake -DCMAKE_BUILD_TYPE=Release ..
make -j2

When I am trying to run make -j2 - I get an error:

make: *** No targets specified and no makefile found.  Stop.

I've inspected all the directories in the build folder but there is no Makefile also. Did the installation process changed?

FUT, 2016-04-05:
Yes, you are totally right. Sorry, closing the issue.

closed 2016-04-05 by FUT