3dtof/voxelsdk #72: windows build instruction error

sravanM, 2016-04-27:
Some clarifications to the wiki instructions that might help other people:

cyapi.h and cyapi.lib should be in voxel-sdk/Voxel/CyAPI and not voxel-sdk/Voxel/

It's unclear whether QT is a requirement, but here's the link to the version you guys are using:
http://download.qt.io/archive/qt/5.4/5.4.0/

Visual studio 2013 express:
https://www.microsoft.com/en-us/download/details.aspx?id=48131

hlprasu, 2016-04-27:
1. Fixed wiki with correct path for CyAPI copying.
2. Qt is a not needed (it might be implicitly used by PCL whenever present
but not necessary).
3. Visual studio 2013 or better is fine. No need to restrict to 2013 alone.

sravanM, 2016-04-27:
re: 2: It complained about QT for me so so perhaps there is some option to not build something that is using it?

3: I think the Boost library in that version of PCL is specific to Visual studio 12 (lib64-msvc-12), it didn't work for me out of the box with 15, but I suppose I could have built boost myself

gadiyar, 2016-04-27:
Also, for VS2013‎, make sure Update 4 or later is installed.

From: Sravan Mettupalli
Sent: Wednesday, April 27, 2016 03:33
To: 3dtof/voxelsdk
Reply To: 3dtof/voxelsdk
Subject: Re: [3dtof/voxelsdk] windows build instruction error (#72)

re: 2: It complained about QT for me so so perhaps there is some option to not build something that is using it?

3: I think the Boost library in that version of PCL is specific to Visual studio 12 (lib64-msvc-12), it didn't work for me out of the box with 15, but I suppose I could have built boost myself

closed 2016-04-28 by sravanM