3dtof/voxelsdk #73: How to use voxelsdk for android platform

tryBetter, 2016-04-28:
I'm trying to use 3dtof for my android TV application. What should I do? Thanks for your anwser.

hlprasu, 2016-04-28:
It is not directly supported. However, there have been previous successful
attempts.

One on issue tracker - https://github.com/3dtof/voxelsdk/issues/45

If you want to built for native Android:
1. Get necessary dependencies ready for Android. If you need PCL, see
https://github.com/PointCloudLibrary/pcl/wiki/Android. If PCL is not needed
then you can comment out add_subdirectory(VoxelPCL) and
add_subdirectory(Apps) in the main CMakeLists.txt. This would mean thst the
Apps will not be built. But several simple programs in Tests/ will still be
built.
2. Second thing to do is to get cmake work with android NDK. See
https://github.com/taka-no-me/android-cmake which could help in this
context.
3. Once your have libraries (.so) built for Android and you are able run
simple Voxel SDK (non-GUI) programs to capture, etc.  from adb shell, next
step is to write a NDK native (Java) wrapper for your app which invokes SDK
API.

This is unofficial and you may need to put some effort to get things in
place.

tryBetter, 2016-04-28:
Thanks for replying soon,I will try it.

tryBetter, 2016-06-01:
help ! I got a problem .
 
`if(!depthCamera->isInitialized())`

`{std::cerr << "Depth camera not initialized for device " << toConnect->id() << std::endl;`

`return env->NewStringUTF("Depth camera not initialized for device ");`

`}`

I can not initialize Depth camera.
Please give me some help., thank you.

tryBetter, 2016-06-01:
My App was run on Android L.

AlexanderSilvaB, 2016-06-01:
@tryBetter , I was having this error too, in my case it was the previleges to access the usb port, I solve it by modifying boot.img and giving root access to all users on USB port, then I flashed this new image and everything works like a charm. To edit boot.img I've used a tool called bootimg.exe, there are tutorials for doing this on internet. Almost forgot, it was needed to change libusb to a specific version and also in android L I need to set the Permissive mode using the setenforce command.

tryBetter, 2016-06-02:
@AlexanderSilvaB , Thank you ! It workes for me . Change my usb dev permission to '0666' and setenforce .

flyflyfly001, 2017-09-08:
@tryBetter Hi，I'm trying to use voxelsdk on android platform, but don't kow how to get started. Could you give me some instructions or compiled voxelsdk for android? Thank you very much!

altheahsu, 2018-02-09:
@tryBetter Hi, Hi，I'm trying to use voxelsdk on android platform, but don't kow how to get started. Could you give me some instructions or compiled voxelsdk for android? Thank you very much!