3dtof/voxelsdk #74: OPT8320 - Voxel 0.6.1 could not find compatible devices

SankethR, 2016-05-04:
Hello,
I am trying to get the OPT8320 up and running after compiling from the source code as given in the Wiki. I have set the Voxel_SDK_PATH and also the Environment variable as given to run the example SimpleVoxelViewer.exe, but the program exits with code -1. (No compatible device found).
I tried to run using both release and debug solution configuration, but the end result is the same. 

What am I doing incorrectly? Any ideas? Thanks...

gadiyar, 2016-05-04:
Do you have an OPT8320 board already?
Are you running SimpleVoxelViewer under Linux or Windows?
Does the pre-built VoxelViewer work with your board?

SankethR, 2016-05-04:
Hello! I am running this under Windows 7 and VS2013 Update 5. 
I already have a OPT8320 board connected, but I am not aware of this pre-built VoxelViewer that you mention. Is there a link to download it? 
Thanks again...

gadiyar, 2016-05-04:
Could you please send me an email at gadiyar (at) ti.com from your official email ID?

Very few OPT8320 boards are in the wild and I'd like to make sure you have the pre-release VoxelViewer to use with this board.

The public one should also work, but is based off voxel-sdk-0.6.0, which has only initial support for OPT8320.

SankethR, 2016-05-04:
I have sent a mail to you. :+1:

SankethR, 2016-05-09:
The board is now up and running with Voxel SDK 0.6.3 (VS2013 Update5, x64 RelWithDebugInfo configuration) after running a Batch script with all the path variables set before executing the start /w C:/.../Voxel.sln.
Thanks for the expert help.

closed 2016-05-09 by SankethR