3dtof/voxelsdk #78: Voxel SDK User Guide example problem

jorjin-image, 2016-05-18:
Hi,
In Voxel SDK User Guide, there is one example “DepthCapture”.
I build this program successfully.
I run the “DepthCapture.exe” with command :
DepthCapture.exe -v 0451 -p 9107 -s 0001 -f testimg -n 1 -t raw

It showed : No valid device found for the specified VID:PID:serialnumber
How can I solve it? Thanks.

gadiyar, 2016-05-18:
Please confirm if the "SimpleVoxelViewer" or "VoxelCLI" programs from the SDK or the VoxelViewer program distributed separately are able to see the camera.

Please confirm that the VOXEL_SDK_PATH environment variable is set correctly.

jorjin-image, 2016-05-19:
The OPT8320 CDK can show images in Voxel Viewer-0.7.16.
But the "SimpleVoxelViewer" and "VoxelCLI" programs have some problems.

In \voxelsdk\App :
Run SimpleVoxelViewer.cmd : It flashed a windows.
[image]
Run VoxelCLIStart.cmd :
[image]

In \voxelsdk\build\bin :
Run SimpleVoxelViewer.exe : 
[image]
Run VoxelCLI.exe : It shows "The procedure entry point could not be located"
[image]

And this is my VOXEL_SDK_PATH environment variable.
[image]

hlprasu, 2016-05-19:
Try C:\Users\Jorjin\voxelsdk\build for VOXEL_SDK_PATH

~ Prasad Bhat

jorjin-image, 2016-05-19:
I changed VOXEL_SDK_PATH to C:\Users\Jorjin\voxelsdk\build.
But it's not working.

jorjin-image, 2016-05-19:
Hello,
I changed VOXEL_SDK_PATH to another voxelsdk(C:\Program Files\Voxel SDK 0.6.3).
C:\Program Files\Voxel SDK 0.6.3 is from https://github.com/3dtof/voxelsdk/wiki/Installation-on-Windows.
C:\Users\Jorjin\voxelsdk\build is from https://github.com/3dtof/voxelsdk/wiki/Build-from-source-on-Windows.
And the "SimpleVoxelViewer" or "VoxelCLI" programs can work.

I used USBDeview to find out VID, PID and serial number.
[image]
Then I run the “DepthCapture.exe” with command :
DepthCapture.exe -v 0451 -p 9107 -s Serial_No._Placeholder -f testimg -n 1 -t raw
But  the “DepthCapture.exe” can't work.
[image]

hlprasu, 2016-05-19:
From the last screenshot, it seems like it has run properly and saved one
frame.

~ Prasad Bhat

jorjin-image, 2016-05-19:
I got the data like following.
[image]
How can I open it?

padi-jagadeesh, 2016-12-18:
Hi,
The file didn't get written as (in the CLI) there is an error saying 
"ERROR:HardwareSerializer: Invalid Config data in Hardware."
I encountered the similar error lately. I checked all the config files which are fine.
Is there anything missing??

I have written my own script in linux which is working fine but this example is well written with more functionality and it would be helpful if it works. 

[image]

gadiyar, 2016-12-19:
@padi-jagadeesh This is just a warning saying there is no pre-calibrated profile saved in the EEPROM.
You can ignore this warning for now.

What is the actual problem you are facing?

padi-jagadeesh, 2016-12-19:
@gadiyar 
Ok.....But, I'm unable to open the dump file......I thought it is binary encoding, so used ".txt"......but not worked....
Is there any format for dumpfile that i need to follow???

gadiyar, 2016-12-19:
Data captured using the DepthCapture.cpp example are just raw binary data. Please read them in your program as raw binary.
Or use python+numpy ( np.fromfile() ) or similar functions in Matlab.

You can refer to the DepthCapture.cpp source code to see the format in which the data is written, and decode accordingly.

padi-jagadeesh, 2016-12-19:
@gadiyar 
Thank you....I have written a C file and read the data now......

If someone needs the decoding format, make use of https://e2e.ti.com/support/sensor/optical_sensors/f/989/t/503049
bin file encoding contains device _id(4 bytes) and Timestamp(8 bytes) in the starting of each frame........consider them while decoding

gadiyar, 2017-02-09:
Closing, please reopen a new issue if this is still a problem.

closed 2017-02-09 by gadiyar