3dtof/voxelsdk #82: Downloader class not working with HaddockCDK

AlexanderSilvaB, 2016-06-01:
Hi,

I'm trying to use the class Downloader with HaddockCDK but everytime I call the download method and it internally calls _usbIO->controlTransfer, this method always return false and I'm presented with the error "USBIO: Control transfer issue.".

Do you know if there's something I can do solve this?
Thank you very much in advance.

gadiyar, 2016-06-02:
Hi Alexander,

We do not support Downloader for Haddock CDK in the Voxel SDK.

For Haddock CDK, firmware upgrades can be done using the 3D Explorer software that originally shipped with the Haddock CDK board.

- Anand

gadiyar, 2016-06-02:
For further support on the Haddock CDK not related to the Voxel SDK, please reach out to us at ti-3dtof (at) list.ti.com.

AlexanderSilvaB, 2016-06-02:
Hi @gadiyar , thank you very much!

It's possible to get the firmware source code to do modifications?  If yes, how can I get it?

Thank you!

gadiyar, 2016-06-02:
Hi @AlexanderSilvaB,

I'm afraid not - we do not share the firmware source and it is not updated frequently.

AlexanderSilvaB, 2016-06-02:
Oh, it's ok. Thank you @gadiyar! :)

closed 2016-06-02 by AlexanderSilvaB