3dtof/voxelsdk #83: Bulk Transfer Failed when capturing frames

PaulDupiano, 2016-06-27:
Hi, running voxel SDK on Ubuntu 14.04 on beglebone black. Trying to run the simple people tracking code found in the demos (https://github.com/3dtof/DemoApplications/tree/master/TinTin/simple_people_tracking)

When I try to run the code I get: ERROR: USBIO: Bulk transfer failed.
I initially thought this was an issue with the sample code, but I tried the other utilities in the voxelsdk/bin like CameraSystemTest, VoxelCLI, and SimpleVoxelViewer and receive similar errors for example:

> root@arm:/home/ubuntu/Software/voxelsdk/build/bin# ./CameraSystemTest -v 0451 -p 9105 -s 922498916191 -f test -n 10
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libvoxel14.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libvoxeld.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/librts5825camera.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libtintincdk.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libhaddockcdk.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libcalculuscdk.so.0.6.3
Detected devices: 
0::0451:9105::922498916191
INFO: TintinCDKCamera: Board revision = 2.0
INFO: ParameterDMLParser: Found register map with name 'OPT9221' (vPG1P0)
INFO: HardwareSerializer: Received 47655 bytes from hardware in 2.14997 s
INFO: Data2DCodec: Original number of 8-bit offsets = 15189
INFO: Data2DCodec: Number of invalid pixels in original data = 0
INFO: Data2DCodec: Current number of 8-bit offsets = 5000
INFO: CameraSystem: Successfully refreshed parameters for TintinCDKCamera(0::0451:9105::922498916191).
Successfully loaded depth camera for device 0::0451:9105::922498916191
INFO: ToFCameraBase: Starting with 160x120@30.0001fps
_dealiasedPhaseMask = 2, _dealiasedPhaseMaskInPhaseOffset = 4
INFO: Capturing at a frame rate of 30.0001 fps
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
WARNING: USBBulkStreamer: Resetting bulk endpoint at 18446744072193809783
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
WARNING: USBBulkStreamer: Resetting bulk endpoint at 18446744072194012911
ERROR: USBIO: Bulk transfer failed.

The error continues on for quite a while then ends with:

> 
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
WARNING: USBBulkStreamer: Resetting bulk endpoint at 2131457746
ERROR: USBBulkStreamer: Did not get a frame in 100 attempts

I initially thought this could have been a permissions problem, and ran it as root, still the same issue. I'm not sure if I'm hitting an issue because I'm using the beaglebone. Any ideas or help in troubleshooting the issue would be helpful

thanks,
paul

larrylisky, 2016-06-27:
Please make sure there is nothing else on the BBB USB but only the CDK. Also don't use HDMI.  Try use VNC to terminal into BBB. 

- Larry

PaulDupiano, 2016-06-28:
Thanks for the response Larry, no HDMI and ran lsusb -t to show I only have the opt8241 evm kit connected, but still got the same error. As a quick sanity check in case there was something wrong with the usb port on the beagle bone, I attached a usb drive and had no problems mounting, reading/writing etc..  Is there anything else you can think of for me to check? 

thanks, 
paul

> root@arm:/home/ubuntu/Software/voxelsdk/build/bin# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=musb-hdrc/1p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=musb-hdrc/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=(Defined at Interface level), Driver=, 480M
root@arm:/home/ubuntu/Software/voxelsdk/build/bin# ./CameraSystemTest -v 0451 -p 9105 -s 922498916191 -f temp -n 1
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libvoxel14.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libvoxeld.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/librts5825camera.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libtintincdk.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libhaddockcdk.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libcalculuscdk.so.0.6.3
Detected devices:
0::0451:9105::922498916191
INFO: TintinCDKCamera: Board revision = 2.0
INFO: ParameterDMLParser: Found register map with name 'OPT9221' (vPG1P0)
INFO: HardwareSerializer: Received 47655 bytes from hardware in 2.15011 s
INFO: Data2DCodec: Original number of 8-bit offsets = 15189
INFO: Data2DCodec: Number of invalid pixels in original data = 0
INFO: Data2DCodec: Current number of 8-bit offsets = 5000
INFO: CameraSystem: Successfully refreshed parameters for TintinCDKCamera(0::0451:9105::922498916191).
Successfully loaded depth camera for device 0::0451:9105::922498916191
INFO: ToFCameraBase: Starting with 320x240@30.0001fps
_dealiasedPhaseMask = 2, _dealiasedPhaseMaskInPhaseOffset = 2
INFO: Capturing at a frame rate of 30.0001 fps
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
WARNING: USBBulkStreamer: Resetting bulk endpoint at 379723381
ERROR: USBIO: Bulk transfer failed.

larrylisky, 2016-06-28:
Next please try run at lower frame size.  Please configure the CDK for 80x60 and see if that works. 

-Larry

PaulDupiano, 2016-06-28:
Hi Larry,

Just tried with both 80x60@30 and 40x30@30 and ran into the same bulk transfer error. Also tried turning down the FPS to 15, no luck there as well

40x30@30fps
> ubuntu@arm:~/Software/voxelsdk/build/bin$ ./CameraSystemTest -v 0451 -p 9105 -s 922498916191 -f ptest -n 1
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libvoxel14.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libvoxeld.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/librts5825camera.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libtintincdk.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libhaddockcdk.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libcalculuscdk.so.0.6.3
Detected devices:
0::0451:9105::922498916191
INFO: TintinCDKCamera: Board revision = 2.0
INFO: ParameterDMLParser: Found register map with name 'OPT9221' (vPG1P0)
INFO: HardwareSerializer: Received 47655 bytes from hardware in 2.1502 s
INFO: Data2DCodec: Original number of 8-bit offsets = 15189
INFO: Data2DCodec: Number of invalid pixels in original data = 0
INFO: Data2DCodec: Current number of 8-bit offsets = 5000
INFO: CameraSystem: Successfully refreshed parameters for TintinCDKCamera(0::0451:9105::922498916191).
Successfully loaded depth camera for device 0::0451:9105::922498916191
INFO: ToFCameraBase: Starting with 40x30@30.0001fps
_dealiasedPhaseMask = 2, _dealiasedPhaseMaskInPhaseOffset = 2
INFO: Capturing at a frame rate of 30.0001 fps
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
WARNING: USBBulkStreamer: Resetting bulk endpoint at 18446744072887180504
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.

80x60@30fps
> ubuntu@arm:~/Software/voxelsdk/build/bin$ ./CameraSystemTest  -v 0451 -p 9105 -s 922498916191 -f temp -n 1
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libvoxel14.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libvoxeld.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/librts5825camera.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libtintincdk.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libhaddockcdk.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libcalculuscdk.so.0.6.3
Detected devices:
0::0451:9105::922498916191
INFO: TintinCDKCamera: Board revision = 2.0
INFO: ParameterDMLParser: Found register map with name 'OPT9221' (vPG1P0)
INFO: HardwareSerializer: Received 47655 bytes from hardware in 2.15008 s
INFO: Data2DCodec: Original number of 8-bit offsets = 15189
INFO: Data2DCodec: Number of invalid pixels in original data = 0
INFO: Data2DCodec: Current number of 8-bit offsets = 5000
INFO: CameraSystem: Successfully refreshed parameters for TintinCDKCamera(0::0451:9105::922498916191).
Successfully loaded depth camera for device 0::0451:9105::922498916191
INFO: ToFCameraBase: Starting with 80x60@30.0001fps
_dealiasedPhaseMask = 2, _dealiasedPhaseMaskInPhaseOffset = 2
INFO: Capturing at a frame rate of 30.0001 fps
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
WARNING: USBBulkStreamer: Resetting bulk endpoint at 18446744071946372683
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.

40x30@15fps
> ubuntu@arm:~/Software/voxelsdk/build/bin$ ./CameraSystemTest -v 0451 -p 9105 -s 922498916191 -f ptest -n 1
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libvoxel14.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libvoxeld.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/librts5825camera.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libtintincdk.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libhaddockcdk.so.0.6.3
INFO: CameraSystem: Successfully loaded factory from library /usr/lib/voxel/libcalculuscdk.so.0.6.3
Detected devices:
0::0451:9105::922498916191
INFO: TintinCDKCamera: Board revision = 2.0
INFO: ParameterDMLParser: Found register map with name 'OPT9221' (vPG1P0)
INFO: HardwareSerializer: Received 47655 bytes from hardware in 2.15033 s
INFO: Data2DCodec: Original number of 8-bit offsets = 15189
INFO: Data2DCodec: Number of invalid pixels in original data = 0
INFO: Data2DCodec: Current number of 8-bit offsets = 5000
INFO: CameraSystem: Successfully refreshed parameters for TintinCDKCamera(0::0451:9105::922498916191).
Successfully loaded depth camera for device 0::0451:9105::922498916191
INFO: ToFCameraBase: Starting with 40x30@15fps
_dealiasedPhaseMask = 2, _dealiasedPhaseMaskInPhaseOffset = 2
INFO: Capturing at a frame rate of 15 fps
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
WARNING: USBBulkStreamer: Resetting bulk endpoint at 18446744073214412369
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.

larrylisky, 2016-06-28:
I think there is some USB issue with BBB.  If you are looking for ARM board that works I had luck with Pandaboard (OMAP4460) and Pico board (AM437x) over USB. 

-Larry

PaulDupiano, 2016-06-28:
Hey Larry, Thanks for the recommendation, and do you mean the Rico Board ? (http://www.myirtech.com/list.asp?id=510)

I've actually got a Rico Board coming my way, and wanted to know if you had any debian or ubuntu images for it.

thanks,
paul

larrylisky, 2016-06-28:
Yes.  Please consult this TI design:

http://www.ti.com/tool/TIDA-00750

Larry

PaulDupiano, 2016-06-28:
Thanks Larry, closing the issue. Probably a USB problem with the beagle bone

closed 2016-06-28 by PaulDupiano