3dtof/voxelsdk #84: Error when building on Ubuntu

widdakay, 2016-07-13:
I have been getting error `/voxelsdk/Voxel/SWIG/../Device.h:63: Error: Syntax error in input(3).` when compiling under Ubuntu. This is in a Docker container to make sure that I have all dependencies listed and it is reproduceable. Here is the Dockerfile:
```Dockerfile
FROM ubuntu:14.04

# Install PCL and other dependencies
RUN apt-get update && apt-get install -y software-properties-common
RUN add-apt-repository ppa:v-launchpad-jochen-sprickerhof-de/pcl
RUN apt-get update && apt-get install -y libpcl-all build-essential git wget tar

# Install updated cmake
RUN wget https://cmake.org/files/v3.5/cmake-3.5.2.tar.gz
RUN tar xf cmake-3.5.2.tar.gz
RUN cd cmake-3.5.2 && ./configure && make -j8 && make install

# Install VTK
RUN git clone git://vtk.org/VTK.git VTK
RUN mkdir /VTK/build && cd /VTK/build && cmake -DCMAKE_BUILD_TYPE:STRING=Release .. && make -j8

# Install VoxelSDK
RUN git clone https://github.com/3dtof/voxelsdk.git
RUN apt-get install -y swig libpcap-dev tcl-vtk python python-dev libudev-dev python-vtk libvtk-java
RUN cd voxelsdk && ./build-all-deb.sh 0.6.3
```

I get the error
```make
Scanning dependencies of target voxelpcl
[ 51%] Building CXX object VoxelPCL/CMakeFiles/voxelpcl.dir/PCLGrabber.cpp.o
/voxelsdk/Voxel/SWIG/../Device.h:63: Error: Syntax error in input(3).
make[2]: *** [lib/python2.7/VoxelPYTHON_wrap.cxx] Error 1
make[1]: *** [Voxel/SWIG/CMakeFiles/_Voxel.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

I am not sure if this is due to compiler versions. Here is the gcc version (should be latest in apt repos):
```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.8/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.8.4-2ubuntu1~14.04.3' --with-bugurl=file:///usr/share/doc/gcc-4.8/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.8 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.8 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libmudflap --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.8-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3) 
```

gadiyar, 2016-07-13:
Could you please confirm the swig version and did you install it from the hlprasu/swig-trusty-backports PPA as mentioned in the wiki?

widdakay, 2016-07-13:
It was a swig version problem. I had only seen the Installation on Linux page and not noticed the other installation page. When I combined them, I was able to get the install working. I ended up with this Dockerfile:

```Dockerfile
FROM ubuntu:14.04

# Install PCL and other dependencies
RUN apt-get update && apt-get install -y software-properties-common
RUN add-apt-repository ppa:hlprasu/swig-trusty-backports
RUN add-apt-repository ppa:v-launchpad-jochen-sprickerhof-de/pcl
RUN apt-get update && apt-get install -y build-essential git wget tar doxygen graphviz dpkg-dev libusb-1.0-0-dev libudev-dev linux-libc-dev swig3.0* python-dev libpcl-all
RUN ln -s /usr/bin/swig{3.0,}

# Install updated cmake
RUN wget https://cmake.org/files/v3.5/cmake-3.5.2.tar.gz
RUN tar xf cmake-3.5.2.tar.gz
RUN cd cmake-3.5.2 && ./configure && make -j8 && make install

# Install VoxelSDK
RUN git clone https://github.com/3dtof/voxelsdk.git
RUN cd voxelsdk && ./build-all-deb.sh 0.6.3
```

closed 2016-07-13 by widdakay