3dtof/voxelsdk #89: Linux : Errors while running make -j2 comaand

ashokbugude, 2016-08-03:
[ 50%] Linking CXX shared library ../lib/libvoxel.so
[ 50%] Built target voxel
[ 51%] Swig source
Scanning dependencies of target voxelpcl
/home/zayyon/Ashok/voxelsdk-0.6.3/Voxel/SWIG/standard_container.i:120: Error: Macro '%std_nodefconst_type' redefined,
/usr/share/swig3.0/std/std_container.i:135: Error: previous definition of '%std_nodefconst_type'.
Voxel/SWIG/CMakeFiles/_Voxel.dir/build.make:61: recipe for target 'lib/python2.7/VoxelPYTHON_wrap.cxx' failed
make[2]: *** [lib/python2.7/VoxelPYTHON_wrap.cxx] Error 1
CMakeFiles/Makefile2:174: recipe for target 'Voxel/SWIG/CMakeFiles/_Voxel.dir/all' failed
make[1]: *** [Voxel/SWIG/CMakeFiles/_Voxel.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 52%] Building CXX object VoxelPCL/CMakeFiles/voxelpcl.dir/PCLGrabber.cpp.o
[ 53%] Linking CXX shared library ../lib/libvoxelpcl.so
[ 53%] Built target voxelpcl
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

larrylisky, 2016-09-04:
It looks like you're missing swig3.0 library.  Follow the below instruction which is also in the voxelsdk wiki:

sudo apt-add-repository ppa:hlprasu/swig-trusty-backports
sudo apt-get install swig3.0* python-dev
sudo ln -s /usr/bin/swig{3.0,}

closed 2016-09-04 by larrylisky

ashokbugude, 2016-09-04:
Hi I have followed those instructions and have swig 3 installed

ashokbugude, 2016-09-04:
It says the macro 'std_nodefconst_type' is redefined as described in error message