3dtof/voxelsdk #90: Linking againt voxel framework

christophk86, 2016-08-12:
Hi everybody,

I am going to build my own project (on windows) which has to be linked againt voxelsdk.
So first step was to build the voxelsdk. Everything works successful.
I configure my cmake file:
set(VOXEL_DIR ../TI_voxel_sdk)
set(VOXEL_Build_DIR ../TI_voxel_sdk/vs_build)
file(GLOB VOXEL_LIBRARIES ${VOXEL_Build_DIR}/lib/*.lib ${VOXEL_Build_DIR}/lib/voxel/*.lib)
set(VOXEL_INCLUDE_DIRS ${VOXEL_Build_DIR}/Voxel ${VOXEL_DIR}/Voxel ${VOXEL_DIR}/VoxelPCL ${VOXEL_DIR}/TI3DToF ${VOXEL_DIR}/Util)
...
target_link_libraries(DepthTest voxel voxelpcl ws2_32 ${VOXEL_LIBRARIES} )

The compilation in Visual Studio 2015 work fine. But if VS15 try to link against voxelsdk following errors occurs:
1>------ Build started: Project: DepthTest, Configuration: Debug Win32 ------
1>DepthTest.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) void __cdecl Voxel::split(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,char,class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > &)" (__imp_?split@Voxel@@YAXABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@DAAV?$vector@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$allocator@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@@3@@Z) referenced in function _main
...

Has anyboy an idea to solve this issue?

Thank you in advance.
Best regards
Chris

larrylisky, 2016-09-04:
Do you have the right visual c++ redistributable library installed?

christophk86, 2016-09-05:
I have already installed the Visual C++ Redistributable für Visual Studio 2015.

gadiyar, 2017-01-04:
We added a fix in https://github.com/3dtof/voxelsdk/commit/e60dc7bd0cd27ea8f205e05587b3f9b65563eb41 to allow building on VS2015. Could you check the latest master branch again