3dtof/voxelsdk #91: Multiple TinTin cameras

Sten7, 2016-08-15:
**This issue is a bug in Voxel SDK that is affecting VoxelViewer as well. It occures when two and more TinTin camera evaluation kits are connected to the same computer.**

When two TinTin camera evaluation kits are connected to the same computer, the video stream is opened only from one of them independent on which camera was selected in VoxelSDK or VoxelViewer. Simillarly any changes in parameters are performed only at the one camera. It seems that the serial numbers of the cameras and the order in that they are connected to the computer (or found by the VoxelSDK) play imporant role in this issue.

**Reproducing the issue with VoxelViewer:**
1) Take two TinTin camera evaluation kits and connect them through USB to the computer and power supply
2) Open first instance of VoxelViewer
3) Select one of the cameras found - Depending on the selection, the issue can occure right now, seeing a stream from the other camera or:
4) Open the second instance of VoxelViewer
5) Select the second camera - You will see the same stream as before, even if the VoxelViewer shows different serial number of the second camera, the stream is identical as well as the destination of any changes in parameters

**Workaround for VoxelViewer:**
1) Connect first the camera with the higher serial number to the computer and power
2) Open the first instance of VoxelViewer and establish a stream from the camera
3) Connect the second camera with the lower serial number to the computer and power
4) Open the second instance of VoxelViewer, select the camera and establish a stream - Now the VoxelViewer works correctly

**Please remove this bug, as it is causing a lot of troubles when more cameras should be automatically handled using the VoxelSDK. Thank you.**

gadiyar, 2016-11-16:
Confirmed, we can reproduce this issue on Windows.

suramyag1, 2017-06-20:
Fixed using https://github.com/3dtof/voxelsdk/commit/a23d4b44f55287a82d88a38f274f0f1fe898037c

closed 2017-06-20 by gadiyar