3dtof/voxelsdk #92: I2C module on the expansion connector of OPT8320 is not responding

padi-jagadeesh, 2016-08-21:
Hi,
I'm working on OPT8320 and trying to communicate using I2C module on the expansion connector through TM4C123GXL(TIVA micro controller). I tried with both the slave addresses 1011000 (58h) and 1011001 (59h), and tried to read from Register 08h (stand by register). But the slave is not acknowledging in the first place. Finally, I want to read the depth information from the board for one of my applications. Where can it be wrong??? Is there any alternative that i can follow to transfer the depth info to an external micro controller????

larrylisky, 2016-09-05:
There are two I2C bus on the expansion connector, one is master (where OPT8320 is a master), and the other is slave.  Be sure to connect to the slave I2C, (J1 pin 9 is SDA, and pin 13 is SCL).  The MCU side will need to provide 10k pull-ups on both lines.

gadiyar, 2017-01-04:
Closing - this is not a VoxelSDK issue

closed 2017-01-04 by gadiyar