3dtof/voxelsdk #95: Skeleton Tracking with voxelsdk

ashokbugude, 2016-09-04:
Can I please know how to implement skeleton tracking with voxelsdk ?

larrylisky, 2016-09-04:
Look for open source application that uses point cloud and OpenNI/NITE.  You can probably leverage those libraries.  voxelsdk outputs the same types of depth/amplitude and 3d data as Kinect.

ashokbugude, 2016-09-04:
Hi Thanks for your response. 
 The pcl(point cloud library) has openi/nite viewer which works fine with kinect v1 and v2.

I am trying to use voxel for TI sensors which does not work with openi/nite as they work only with kinect.
 Could I please know how to use voxel or any other library to implement skeleton tracking with Ti's TOF sensors

larrylisky, 2016-09-05:
@ashokbugude:  You may want to try this:

YouTube:
[https://www.youtube.com/watch?v=FmXbS5DkSsw](url)
Code:
[https://github.com/Norman0406/OpenSkeletonFitting](url)

Look in the OpenSFInput folder, and follow the InputKinect class example to create a new input class InputVoxel by deriving from the Input class.

SneakyLuke88, 2016-09-19:
Here is another solution, base on TI sensors, with fully compatible with OpenNI V1.5 and 2.2
https://www.youtube.com/watch?v=_qrw2AhSthI

SneakyLuke88, 2016-10-22:
You could use this 
http://www.lips-hci.com/products/sdk/

gadiyar, 2017-02-09:
Closing, this is not a voxelsdk issue.

closed 2017-02-09 by gadiyar