3dtof/voxelsdk #96: libvoxelpcl.so not instaslled

ashokbugude, 2016-10-03:
I have installed voxelsdk in Linux 16.04 but 'libvoxelpcl.so ' is not installed . It is needed to run 'hand tracking ' application of https://github.com/3dtof/DemoApplications/tree/master/TinTin/hand_tracking

I get the below error messages when I run make command

zayyon@zayyon-Lenovo-ideapad-300-15ISK:~/Downloads/DemoApplications-master/TinTin/hand_tracking/build$ sudo make
[sudo] password for zayyon: 
make[2]: *** No rule to make target '/usr/lib/libvoxelpcl.so', needed by 'main'.  Stop.
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/main.dir/all' failed
make[1]: *** [CMakeFiles/main.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

Could you please send me a copy of 64 bit libvoxelpcl.so file
Thanks

larrylisky, 2016-10-03:
Have you installed voxelsdk?

libvoxelpcl.so should be built and installed when you install voxelsdk.

-Larry

ashokbugude, 2016-10-03:
I have installed voxel sdk , but have only files related to voxel which does not include libvoxelpcl as show below
[image]

Could you pls send libvoxelpcl.so file

ashokbugude, 2016-10-07:
The issue was due to https://github.com/3dtof/voxelsdk/issues/89

closed 2016-10-07 by ashokbugude