3dtof/voxelsdk #97: Windows Install Error needing Python27

lynetta, 2016-10-11:
I am trying to install Voxel-SDK-0.6.0-win64-release.exe with PCL-1.7.2-AllInOne-win64.exe and qt-opensource-windows-x86-msvc2013_64-5.4.2.exe.  I get an error while installing the SDK looking for Python27.  Python27 is installed at C:\Python27.  I added this to my PATH but that still didn't seem to resolve the issue.  I unchecked the Voxel Python Bindings from the "Choose Components" list.  Can you verify that this is "ok"?  Why is it failing?  I have Python 2.7.12 installed.

gadiyar, 2016-10-11:
Check if the Python version you have is 32-bit or 64-bit.

The 64-bit version will give you the following message at startup (note "64 bit (AMD64)"):
Python 2.7.12 (v2.7.12:d33e0cf91556, Jun 27 2016, 15:24:40) [MSC v.1500 64 bit (AMD64)] on win32

and can be downloaded from here:
https://www.python.org/ftp/python/2.7.12/python-2.7.12.amd64.msi

gadiyar, 2016-10-11:
Also, it is okay to uncheck the Python bindings - however this will not allow you to use the VoxelSDK you built with VoxelViewer.

lynetta, 2016-10-11:
Thanks so much.  With the 64bit Python things worked well.  Now the real fun begins!

closed 2016-10-11 by lynetta