3dtof/voxelsdk #98: libvoxel.so.34: undefined symbol: udev_new

OhSeoHwan, 2016-11-07:
ubuntu 64bit LTS

./SimpleVoxelViewer
INFO: CameraSystem: Successfully loaded factory from library /voxelsdk-0.6.5/build/lib/voxel/librts5825camera.so.0.6.5
INFO: CameraSystem: Successfully loaded factory from library /voxelsdk-0.6.5/build/lib/voxel/libvoxeld.so.0.6.5
INFO: CameraSystem: Successfully loaded factory from library /voxelsdk-0.6.5/build/lib/voxel/libtintincdk.so.0.6.5
INFO: CameraSystem: Successfully loaded factory from library /voxelsdk-0.6.5/build/lib/voxel/libvoxel14.so.0.6.5
INFO: CameraSystem: Successfully loaded factory from library /voxelsdk-0.6.5/build/lib/voxel/libcalculuscdk.so.0.6.5
INFO: CameraSystem: Successfully loaded factory from library /voxelsdk-0.6.5/build/lib/voxel/libhaddockcdk.so.0.6.5
./SimpleVoxelViewer: symbol lookup error: /01.Project/10.TI-Test/01.TEST/lib/libvoxel.so.34: undefined symbol: udev_new

How to solve it? /libvoxel.so.34: undefined symbol: udev_new

gadiyar, 2016-11-07:
Which Ubuntu version is this? And do you have libudev-dev installed? (Run "dpkg --get-selections *udev*" to check)

OhSeoHwan, 2016-11-07:
My OS status.
#lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.4 LTS
Release:	14.04
Codename:	trusty

-------------------
installed libudev-dev
# dpkg --get-selections udev
udev						install

gadiyar, 2016-11-07:
Sorry, please run `dpkg --get-selections *udev*` - markdown mangled the previous command

OhSeoHwan, 2016-11-07:
:dpkg --get-selections *udev*
gir1.2-gudev-1.0				install
libgudev-1.0-0:amd64				install
libudev-dev					install
libudev1:amd64					install
libudev1:i386					install
system-config-printer-udev			install
udev						install

gadiyar, 2016-11-07:
It looks like you have the i386 version of libudev installed for some reason. I'm not sure if you're accidentally linking to that instead.

Can you run `ldd SimpleVoxelViewer` to check?

OhSeoHwan, 2016-11-07:
ldd SimpleVoxelViewer | grep libudev*
not fount udev

---------------------------------
build/bin# ldd SimpleVoxelViewer 
	linux-vdso.so.1 =>  (0x00007fff64de5000)
	libvoxelpcl.so.34 => /10.TI-Test/01.TEST/voxelsdk-0.6.6/build/lib/libvoxelpcl.so.34 (0x00007f6b74f7b000)
	libboost_system.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.54.0 (0x00007f6b74d77000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f6b74b59000)
	libpcl_common.so.1.7 => /usr/lib/libpcl_common.so.1.7 (0x00007f6b748a5000)
	libpcl_visualization.so.1.7 => /usr/lib/libpcl_visualization.so.1.7 (0x00007f6b7454e000)
	libvtkRendering.so.5.8 => /usr/lib/libvtkRendering.so.5.8 (0x00007f6b73e48000)
	libvtkFiltering.so.5.8 => /usr/lib/libvtkFiltering.so.5.8 (0x00007f6b738bf000)
	libvtkCommon.so.5.8 => /usr/lib/libvtkCommon.so.5.8 (0x00007f6b73349000)
	libvoxel.so.34 => /01.Project/10.TI-Test/01.TEST/voxelsdk-0.6.6/build/lib/libvoxel.so.34 (0x00007f6b72f02000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f6b72bfe000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f6b728f8000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f6b726e2000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f6b7231d000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f6b751df000)
	libboost_thread.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0 (0x00007f6b72107000)
	libgomp.so.1 => /usr/lib/x86_64-linux-gnu/libgomp.so.1 (0x00007f6b71ef8000)
	libboost_filesystem.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.54.0 (0x00007f6b71ce2000)
	libpcl_io.so.1.7 => /usr/lib/libpcl_io.so.1.7 (0x00007f6b7182a000)
	libpcl_kdtree.so.1.7 => /usr/lib/libpcl_kdtree.so.1.7 (0x00007f6b71503000)
	libvtkImaging.so.5.8 => /usr/lib/libvtkImaging.so.5.8 (0x00007f6b70f16000)
	libvtkGraphics.so.5.8 => /usr/lib/libvtkGraphics.so.5.8 (0x00007f6b7055a000)
	libvtkIO.so.5.8 => /usr/lib/libvtkIO.so.5.8 (0x00007f6b6fdd9000)
	libvtkHybrid.so.5.8 => /usr/lib/libvtkHybrid.so.5.8 (0x00007f6b6f8f7000)
	libvtkWidgets.so.5.8 => /usr/lib/libvtkWidgets.so.5.8 (0x00007f6b6f523000)
	libvtkCharts.so.5.8 => /usr/lib/libvtkCharts.so.5.8 (0x00007f6b6f245000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1 (0x00007f6b6efdf000)
	libvtkftgl.so.5.8 => /usr/lib/libvtkftgl.so.5.8 (0x00007f6b6edd4000)
	libQtGui.so.4 => /usr/lib/x86_64-linux-gnu/libQtGui.so.4 (0x00007f6b6e121000)
	libQtCore.so.4 => /usr/lib/x86_64-linux-gnu/libQtCore.so.4 (0x00007f6b6dc3c000)
	libgl2ps.so.0 => /usr/lib/libgl2ps.so.0 (0x00007f6b6da29000)
	libXt.so.6 => /usr/lib/x86_64-linux-gnu/libXt.so.6 (0x00007f6b6d7c3000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f6b6d48e000)
	libvtksys.so.5.8 => /usr/lib/libvtksys.so.5.8 (0x00007f6b6d252000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f6b6cfaf000)
	libusb-1.0.so.0 => /usr/local/lib/libusb-1.0.so.0 (0x00007f6b6cd9f000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f6b6cb9b000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f6b6c993000)
	libboost_iostreams.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_iostreams.so.1.54.0 (0x00007f6b6c77a000)
	libboost_chrono.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_chrono.so.1.54.0 (0x00007f6b6c573000)
	libpcl_io_ply.so.1.7 => /usr/lib/libpcl_io_ply.so.1.7 (0x00007f6b6c324000)
	libpng12.so.0 => /lib/x86_64-linux-gnu/libpng12.so.0 (0x00007f6b6c0fe000)
	libOpenNI2.so.0 => /usr/lib/libOpenNI2.so.0 (0x00007f6b6bea3000)
	libOpenNI.so.0 => /usr/lib/libOpenNI.so.0 (0x00007f6b6bc2b000)
	libvtkverdict.so.5.8 => /usr/lib/libvtkverdict.so.5.8 (0x00007f6b6b9f5000)
	libpq.so.5 => /usr/lib/libpq.so.5 (0x00007f6b6b7c6000)
	libmysqlclient.so.18 => /usr/lib/x86_64-linux-gnu/libmysqlclient.so.18 (0x00007f6b6b28e000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f6b6b075000)
	libvtkDICOMParser.so.5.8 => /usr/lib/libvtkDICOMParser.so.5.8 (0x00007f6b6ae5c000)
	libnetcdf.so.7 => /usr/lib/libnetcdf.so.7 (0x00007f6b6ab1f000)
	libnetcdf_c++.so.4 => /usr/lib/libnetcdf_c++.so.4 (0x00007f6b6a902000)
	libvtkmetaio.so.5.8 => /usr/lib/libvtkmetaio.so.5.8 (0x00007f6b6a675000)
	libsqlite3.so.0 => /usr/lib/x86_64-linux-gnu/libsqlite3.so.0 (0x00007f6b6a3bc000)
	libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007f6b6a167000)
	libtiff.so.5 => /usr/lib/x86_64-linux-gnu/libtiff.so.5 (0x00007f6b69ef5000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f6b69ccb000)
	libavformat.so.54 => /usr/lib/x86_64-linux-gnu/libavformat.so.54 (0x00007f6b699a9000)
	libavcodec.so.54 => /usr/lib/x86_64-linux-gnu/libavcodec.so.54 (0x00007f6b68c54000)
	libavutil.so.52 => /usr/lib/x86_64-linux-gnu/libavutil.so.52 (0x00007f6b68a2f000)
	libswscale.so.2 => /usr/lib/x86_64-linux-gnu/libswscale.so.2 (0x00007f6b687e8000)
	libvtkParallel.so.5.8 => /usr/lib/libvtkParallel.so.5.8 (0x00007f6b683e5000)
	libvtkexoIIc.so.5.8 => /usr/lib/libvtkexoIIc.so.5.8 (0x00007f6b6819b000)
	libvtkViews.so.5.8 => /usr/lib/libvtkViews.so.5.8 (0x00007f6b67f01000)
	libvtkInfovis.so.5.8 => /usr/lib/libvtkInfovis.so.5.8 (0x00007f6b67a69000)
	libglapi.so.0 => /usr/lib/x86_64-linux-gnu/libglapi.so.0 (0x00007f6b67842000)
[... 64 lines cut ...]
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007f6b5df5a000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007f6b5dd3f000)
	liblber-2.4.so.2 => /usr/lib/x86_64-linux-gnu/liblber-2.4.so.2 (0x00007f6b5db30000)
[... cut ...]
	libroken.so.18 => /usr/lib/x86_64-linux-gnu/libroken.so.18 (0x00007f6b5b77c000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f6b5b577000)
	libnuma.so.1 => /usr/lib/x86_64-linux-gnu/libnuma.so.1 (0x00007f6b5b36c000)
	libwind.so.0 => /usr/lib/x86_64-linux-gnu/libwind.so.0 (0x00007f6b5b143000)
	libheimbase.so.1 => /usr/lib/x86_64-linux-gnu/libheimbase.so.1 (0x00007f6b5af35000)
	libhx509.so.5 => /usr/lib/x86_64-linux-gnu/libhx509.so.5 (0x00007f6b5acec000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007f6b5aab3000)

OhSeoHwan, 2016-11-08:
/voxelsdk-0.6.5/build/bin#./SimpleVoxelViewer 
INFO: CameraSystem: Successfully loaded factory from library /voxelsdk-0.6.5/build/lib/voxel/librts5825camera.so.0.6.5
INFO: CameraSystem: Successfully loaded factory from library /voxelsdk-0.6.5/build/lib/voxel/libvoxeld.so.0.6.5
INFO: CameraSystem: Successfully loaded factory from library /voxelsdk-0.6.5/build/lib/voxel/libtintincdk.so.0.6.5
INFO: CameraSystem: Successfully loaded factory from library /voxelsdk-0.6.5/build/lib/voxel/libvoxel14.so.0.6.5
INFO: CameraSystem: Successfully loaded factory from library /voxelsdk-0.6.5/build/lib/voxel/libcalculuscdk.so.0.6.5
INFO: CameraSystem: Successfully loaded factory from library /voxelsdk-0.6.5/build/lib/voxel/libhaddockcdk.so.0.6.5
**# ./SimpleVoxelViewer: symbol lookup error: /voxelsdk-0.6.5/build/lib/libvoxel.so.34: undefined symbol: udev_new**

---------------------------------
# nm libvoxel.so.34 
 U udev_device_get_devnode
 U udev_device_get_parent_with_subsystem_devtype
                 U udev_device_get_sysattr_value
                 U udev_device_new_from_syspath
                 U udev_device_unref
                 U udev_enumerate_add_match_property
                 U udev_enumerate_add_match_subsystem
                 U udev_enumerate_get_list_entry
                 U udev_enumerate_new
                 U udev_enumerate_scan_devices
                 U udev_enumerate_unref
                 U udev_list_entry_get_name
                 U udev_list_entry_get_next
                 U udev_new
                 U udev_unref
-------------------------------------------
libvoxel.so.34  The udev_new symbol appears in the file information. However, I can not refer to udev_new symbol. All example programs have the same behavior.
How can I solve the problem?

gadiyar, 2016-11-08:
Here's what I get:

   user1@ubuntudev:~/voxelsdk-arm/voxelsdk-arm/build/bin$ ldd SimpleVoxelViewer  | grep udev
        libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f31b9d59000)

Just on a hunch, can you uninstall libudev1:i386 and rebuild cleanly.
Also could you save your build logs maybe?

OhSeoHwan, 2016-11-08:
My test command line
-------------------
apt-get remove libudev1:i386
rm -rf voxelsdk-0.6.5
tar -xvf voxelsdk-0.6.5.tar.gz
mkdir build
cd build
export CMAKE_PREFIX_PATH=$PWD:$CMAKE_PREFIX_PATH
cmake -DCMAKE_BUILD_TYPE=Release ..
make -j2
export VOXEL_SDK_PATH=/TI-Test/01.TEST/voxelsdk-0.6.5/build
export PATH=$VOXEL_SDK_PATH/lib:$VOXEL_SDK_PATH/bin:$PATH
export PYTHONPATH=$VOXEL_SDK_PATH/lib/python2.7:$PYTHONPATH
---------------------------------------
Build logs?? What does it mean? make log?

/voxelsdk-0.6.5/build-test# make -j2
Scanning dependencies of target filesToCopyTarget
Scanning dependencies of target voxel
[  1%] Generating ../share/voxel-0.6.5/conf/CalculusCDKCamera.conf
[  2%] Generating ../share/voxel-0.6.5/conf/CalculusCDKCameraLensOnly.conf
[  3%] Generating ../share/voxel-0.6.5/conf/CalculusCDKCameraNormal.conf
[  4%] Generating ../share/voxel-0.6.5/conf/HaddockCDKCamera.conf
[  5%] Building CXX object Voxel/CMakeFiles/voxel.dir/Device.cpp.o
[  6%] Generating ../share/voxel-0.6.5/conf/HaddockCDKCameraNormal.conf
[  7%] Generating ../share/voxel-0.6.5/conf/TintinCDKCamera.conf
[  7%] Generating ../share/voxel-0.6.5/conf/TintinCDKCameraLensOnly.conf
[  8%] Generating ../share/voxel-0.6.5/conf/TintinCDKCameraShortRange.conf
[  9%] Generating ../share/voxel-0.6.5/conf/TintinCDKCameraLongRange.conf
[ 10%] Generating ../share/voxel-0.6.5/conf/TintinCDKCameraHighAmbient.conf
[ 11%] Generating ../share/voxel-0.6.5/conf/TintinCDKCameraNoCalibration.conf
[ 12%] Generating ../share/voxel-0.6.5/conf/TintinCDKCameraPhaseOffset.bin
[ 13%] Generating ../share/voxel-0.6.5/conf/RTS5825Camera.conf
[ 13%] Generating ../share/voxel-0.6.5/conf/RTS5825CameraNormal.conf
[ 14%] Generating ../share/voxel-0.6.5/conf/RTS5825CameraNoCalibration.conf
[ 15%] Generating ../share/voxel-0.6.5/conf/VoxelDCamera.conf
[ 16%] Generating ../share/voxel-0.6.5/conf/VoxelDCameraNormal.conf
[ 17%] Generating ../share/voxel-0.6.5/conf/Voxel14Camera.conf
[ 18%] Generating ../share/voxel-0.6.5/conf/Voxel14CameraNormal.conf
[ 19%] Generating ../share/voxel-0.6.5/conf/Voxel14CameraTestMode.conf
[ 19%] Generating ../share/voxel-0.6.5/conf/Voxel14CameraPhaseOffset.bin
[ 20%] Generating ../share/voxel-0.6.5/conf/OPT8320.dml
[ 21%] Generating ../share/voxel-0.6.5/conf/OPT9220.dml
[ 22%] Generating ../share/voxel-0.6.5/conf/OPT9221.dml
[ 23%] Generating ../share/voxel-0.6.5/fw/OPT9220_0v27.tip
[ 24%] Generating ../share/voxel-0.6.5/fw/OPT9221_0v23.tip
[ 24%] Built target filesToCopyTarget
[ 25%] Building CXX object Voxel/CMakeFiles/voxel.dir/Common.cpp.o
[ 26%] Building CXX object Voxel/CMakeFiles/voxel.dir/DataPacket.cpp.o
[ 26%] Building CXX object Voxel/CMakeFiles/voxel.dir/Downloader.cpp.o
[ 27%] Building CXX object Voxel/CMakeFiles/voxel.dir/DFT.cpp.o
[ 28%] Building CXX object Voxel/CMakeFiles/voxel.dir/Data2DCodec.cpp.o
[ 29%] Building CXX object Voxel/CMakeFiles/voxel.dir/Parameter.cpp.o
[ 30%] Building CXX object Voxel/CMakeFiles/voxel.dir/RegisterProgrammer.cpp.o
[ 31%] Building CXX object Voxel/CMakeFiles/voxel.dir/USBSystem.cpp.o
[ 32%] Building CXX object Voxel/CMakeFiles/voxel.dir/UVC.cpp.o
[ 32%] Building CXX object Voxel/CMakeFiles/voxel.dir/UVCXU.cpp.o
[ 33%] Building CXX object Voxel/CMakeFiles/voxel.dir/Logger.cpp.o
[ 34%] Building CXX object Voxel/CMakeFiles/voxel.dir/Timer.cpp.o
[ 35%] Building CXX object Voxel/CMakeFiles/voxel.dir/Streamer.cpp.o
[ 36%] Building CXX object Voxel/CMakeFiles/voxel.dir/USBIO.cpp.o
[ 37%] Building CXX object Voxel/CMakeFiles/voxel.dir/USBBulkStreamer.cpp.o
[ 38%] Building CXX object Voxel/CMakeFiles/voxel.dir/UVCStreamer.cpp.o
[ 39%] Building CXX object Voxel/CMakeFiles/voxel.dir/HardwareSerializer.cpp.o
[ 39%] Building CXX object Voxel/CMakeFiles/voxel.dir/CameraSystem.cpp.o
[ 40%] Building CXX object Voxel/CMakeFiles/voxel.dir/DepthCamera.cpp.o
[ 41%] Building CXX object Voxel/CMakeFiles/voxel.dir/FrameStream.cpp.o
[ 42%] Building CXX object Voxel/CMakeFiles/voxel.dir/FrameGenerator.cpp.o
[ 43%] Building CXX object Voxel/CMakeFiles/voxel.dir/DepthCameraLibrary.cpp.o
[ 44%] Building CXX object Voxel/CMakeFiles/voxel.dir/TinyXML2.cpp.o
[... 141 lines cut ...]
Linking CXX shared library ../../../lib/voxel/libhaddockcdk.so
[ 90%] Built target haddockcdk
Scanning dependencies of target rts5825camera
[ 91%] Building CXX object TI3DToF/boards/RTS5825Camera/CMakeFiles/rts5825camera.dir/RTS5825Camera.cpp.o
[ 92%] Building CXX object TI3DToF/boards/RTS5825Camera/CMakeFiles/rts5825camera.dir/RTS5825CameraFactory.cpp.o
[ 93%] Building CXX object TI3DToF/boards/TintinCDK/CMakeFiles/tintincdk.dir/TintinCDKCameraFactory.cpp.o
Linking CXX shared library ../../../lib/voxel/librts5825camera.so
[ 93%] Built target rts5825camera
[ 94%] Building CXX object TI3DToF/boards/TintinCDK/CMakeFiles/tintincdk.dir/TintinCDKDownloaderFactory.cpp.o
Scanning dependencies of target voxeld
[ 95%] Building CXX object TI3DToF/boards/VoxelD/CMakeFiles/voxeld.dir/VoxelDCamera.cpp.o
[ 96%] Building CXX object TI3DToF/boards/TintinCDK/CMakeFiles/tintincdk.dir/TintinEEPROMDownloader.cpp.o
[ 97%] Building CXX object TI3DToF/boards/VoxelD/CMakeFiles/voxeld.dir/VoxelDCameraFactory.cpp.o
Linking CXX shared library ../../../lib/voxel/libtintincdk.so
[ 97%] Built target tintincdk
Scanning dependencies of target voxel14
[ 98%] Building CXX object TI3DToF/boards/Voxel14/CMakeFiles/voxel14.dir/Voxel14Camera.cpp.o
Linking CXX shared library ../../../lib/voxel/libvoxeld.so
[ 98%] Built target voxeld
[100%] Building CXX object TI3DToF/boards/Voxel14/CMakeFiles/voxel14.dir/Voxel14CameraFactory.cpp.o
Linking CXX shared library ../../../lib/voxel/libvoxel14.so
[100%] Built target voxel14
---------------------
ldd SimpleVoxelViewer | grep udev
Not found udev....

gadiyar, 2016-11-08:
Could you post the result of this step:
   `cmake -DCMAKE_BUILD_TYPE=Release ..`

OhSeoHwan, 2016-11-08:
[a.txt](https://github.com/3dtof/voxelsdk/files/579611/a.txt)

gadiyar, 2017-01-04:
This was caused by conflicting i386 packages for libudev-dev. Need to check if we can detect this in CMake automatically.