3dtof/voxelsdk #100: Install Voxel SDK 0.6.0 on Windows 10

joel92217, 2016-11-11:
Hi, I try to install Voxel SDK 0.6.0 on WIndows 10.
I have installed Python 2.7.11(64bit), PCL-1.7.2(win64) and Qt 5.4.2 in advance.
The Voxel SDK is successfully installed.
After running Simple Voxel Viewer program, some error messages pop up like "pcl_common_release.dll", "pcl_visualization_release.dll", "vtkRenderingFreeTypeOpenGL-6.1.dll", "vtkRenderingOpenGL-6.1.dll" are lost in my PC.

After install/uninstall voxel sdk many times, I finally try to copy all the PCL 1.7.2 DLL files to the Voxel SDK /bin directory. It works though many warning messages appears. 
 For example,
 WARNING: CameraSystem: Ignoring Voxel library pcl_common_release.dll with ABI version = 0. Expected ABI version = 34
 Suppose the pre-built binaries can not locate the DLL files in PCL library, please help check what's going on.
 After running Simple VoxelViewer, an error message shows no device attached. Program is terminated and command window is closed.
 As a result, I can not confirm the error message exactly. Does Simple VoxelViewer need an optical sensor CDK device to execute normally?

gadiyar, 2016-11-11:
Most tools included in the SDK require a connected CDK or supported 3DTOF camera.
SimpleVoxelViewer attempts to connect to a supported 3DTOF camera at startup and start a frame stream.

Another tool in the SDK (CameraSystemReadStreamTest.exe) can read a pre-captured .vxl files and extract data from it.

To debug your installation problem, please check your PATH and VOXEL_SDK_PATH environment variables.
https://github.com/3dtof/voxelsdk/wiki/Build-from-source-on-Windows#running-sdk-programs

joel92217, 2016-11-11:
Yes, it works after adding the path environment. Should take care of some version numer and path name in practical directory. Please add the environment setting notification in the Wiki Installation on Windows page. Thank you for your help.

gadiyar, 2016-11-11:
Thanks.

Will update the wiki page. I realize it's only on the Build Instructions page.

closed 2016-11-11 by joel92217