3dtof/voxelsdk #108: USBIO error :Bulk transfer failed.       Info is listed below:

aigoze, 2017-01-20:
root@ubuntu:/home/jiandonghuang/Documents/TiDethSensor/3dtof/voxelsdk/build# VoxelCLI 
INFO: CameraSystem: Successfully loaded factory from library /home/jiandonghuang/Documents/TiDethSensor/3dtof/voxelsdk/build/lib/voxel/libhaddockcdk.so.0.6.7
INFO: CameraSystem: Successfully loaded factory from library /home/jiandonghuang/Documents/TiDethSensor/3dtof/voxelsdk/build/lib/voxel/libvoxel14.so.0.6.7
INFO: CameraSystem: Successfully loaded factory from library /home/jiandonghuang/Documents/TiDethSensor/3dtof/voxelsdk/build/lib/voxel/libvoxeld.so.0.6.7
INFO: CameraSystem: Successfully loaded factory from library /home/jiandonghuang/Documents/TiDethSensor/3dtof/voxelsdk/build/lib/voxel/librts5825camera.so.0.6.7
INFO: CameraSystem: Successfully loaded factory from library /home/jiandonghuang/Documents/TiDethSensor/3dtof/voxelsdk/build/lib/voxel/libtintincdk.so.0.6.7
INFO: CameraSystem: Successfully loaded factory from library /home/jiandonghuang/Documents/TiDethSensor/3dtof/voxelsdk/build/lib/voxel/libcalculuscdk.so.0.6.7
INFO: ParameterDMLParser: Found register map with name 'calculus' (vPG2P0)

ERROR: HardwareSerializer: Invalid config data in hardware.
WARNING: MainConfigurationFile: Failed to read configuration from hardware.
WARNING: MainConfigurationFile: Could not open file '885691.bin'
INFO: DepthCamera: Setting parameter 'unambiguous_range'
INFO: DepthCamera: Setting parameter 'mix_volt'
INFO: DepthCamera: Setting parameter 'mod_freq1'
INFO: DepthCamera: Setting parameter 'mod_cdriv_en'
INFO: DepthCamera: Setting parameter 'mod_cdriv_curr'
INFO: DepthCamera: Setting parameter 'sub_frame_cnt_max1'
INFO: DepthCamera: Setting parameter 'intg_time'
INFO: DepthCamera: Setting parameter 'shutter_en'
INFO: DepthCamera: Setting parameter 'high_ambient_en'
INFO: DepthCamera: Setting parameter 'disable_illum_p'
INFO: DepthCamera: Setting parameter 'disable_illum_n'
INFO: DepthCamera: Setting parameter 'pvdd_voltage'
INFO: DepthCamera: Setting parameter 'pvdd_en'
INFO: CameraSystem: Successfully refreshed parameters for CalculusCDKCamera(0::0451:9107::885691).
INFO: ToFCameraBase: Starting with 80x60@60fps
_dealiasedPhaseMask = 0, _dealiasedPhaseMaskInPhaseOffset = 0
voxel:CalculusCDKCamera(0::0451:9107::885691)> ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
WARNING: USBBulkStreamer: Resetting bulk endpoint at 1484873485310147
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
WARNING: USBBulkStreamer: Resetting bulk endpoint at 1484873486360632
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
WARNING: USBBulkStreamer: Resetting bulk endpoint at 1484873487610981
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
WARNING: USBBulkStreamer: Resetting bulk endpoint at 1484873489059644
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
WARNING: USBBulkStreamer: Resetting bulk endpoint at 1484873490309464
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
ERROR: USBIO: Bulk transfer failed.
ERROR: USBBulkStreamer: Bulk End point reset failed
WARNING: USBBulkStreamer: Resetting bulk endpoint at 1484873490695868

^C
root@ubuntu:/home/jiandonghuang/Documents/TiDethSensor/3dtof/voxelsdk/build# VoxelCLI 
INFO: CameraSystem: Successfully loaded factory from library /home/jiandonghuang/Documents/TiDethSensor/3dtof/voxelsdk/build/lib/voxel/libhaddockcdk.so.0.6.7
INFO: CameraSystem: Successfully loaded factory from library /home/jiandonghuang/Documents/TiDethSensor/3dtof/voxelsdk/build/lib/voxel/libvoxel14.so.0.6.7
INFO: CameraSystem: Successfully loaded factory from library /home/jiandonghuang/Documents/TiDethSensor/3dtof/voxelsdk/build/lib/voxel/libvoxeld.so.0.6.7
INFO: CameraSystem: Successfully loaded factory from library /home/jiandonghuang/Documents/TiDethSensor/3dtof/voxelsdk/build/lib/voxel/librts5825camera.so.0.6.7
INFO: CameraSystem: Successfully loaded factory from library /home/jiandonghuang/Documents/TiDethSensor/3dtof/voxelsdk/build/lib/voxel/libtintincdk.so.0.6.7
INFO: CameraSystem: Successfully loaded factory from library /home/jiandonghuang/Documents/TiDethSensor/3dtof/voxelsdk/build/lib/voxel/libcalculuscdk.so.0.6.7
INFO: ParameterDMLParser: Found register map with name 'calculus' (vPG2P0)

ERROR: HardwareSerializer: Invalid config data in hardware.
WARNING: MainConfigurationFile: Failed to read configuration from hardware.
WARNING: MainConfigurationFile: Could not open file '885691.bin'
INFO: CameraSystem: Successfully refreshed parameters for CalculusCDKCamera(0::0451:9107::885691).
INFO: ToFCameraBase: Starting with 80x60@60fps
_dealiasedPhaseMask = 0, _dealiasedPhaseMaskInPhaseOffset = 0
[... 146 lines cut ...]
|     for now until we fix it cleanly)
|  
* commit e6d9d8b8ac6d756072fc252b3f2226218dd9c57b
| Author: Gajanan Maske <gajanan.maske@pathpartnertech.com>
| Date:   Mon Dec 19 16:38:06 2016 +0530
| 
|     MedianFilter: optimizations for ARM
|  
* commit 91a05f2116e4244b7ded231578d61b6e7cdee475
| Author: Gajanan Maske <gajanan.maske@pathpartnertech.com>
| Date:   Mon Dec 19 16:18:22 2016 +0530
| 
|     MedianFilter: some optimizations
|  
* commit 14c666a2a0784fc8876c93ccc6555b37119ecb63
| Author: Gajanan Maske <gajanan.maske@pathpartnertech.com>
| Date:   Mon Dec 19 16:14:20 2016 +0530
| 
|     Some ARM optimizations
|  
---------------------------------------------------------------------------------------------------------------

root@ubuntu:/home/jiandonghuang/Documents/TiDethSensor/3dtof/voxelsdk# git status
On branch master
Your branch is up-to-date with 'origin/master'.

gadiyar, 2017-01-20:
This doesn't look like a software issue.

Does your camera work with VoxelViewer under Windows? If not, your camera might have a hardware issue.

aigoze, 2017-01-20:
The camera board is CalculusCDK and it works fine on Windows platform using VoxelViewer. 
I got these results while running VoxelCLI on my Virtual machine ubuntu. I was wondering if it was the Virtual Machine issue. 
btw, my virtual machine is parllel desptop and my host OS is Mac OS sierra.

gadiyar, 2017-01-20:
Virtual machines could have this type of issue.

aigoze, 2017-02-07:
Finally, I find an computer which run Ubuntu 14.04 rather than a virtual machine. And this problem had been solved by recompiling the PCL library on this Ubuntu. After that, the VoxelCLI can be run without USB warning issues.
But the VoxelViewer on Windows 10 somtimes report this type of issue in the log.

closed 2017-02-09 by aigoze