3dtof/voxelsdk #110: Voxel SDK 0.6.5 prebuilt binaries installer can't find python

Arrowana, 2017-02-08:
Trying to install Voxel SDK Voxel.SDK-0.6.5-win64.exe prebuilt binaries on Windows 10 64-bit.

I get an error message
"Could not locate Python 2.7 installation. This is required to install to Python bindings." 

However Python 2.7 is installed and in my PATH, so I don't get what is wrong

Arrowana, 2017-02-09:
The macro getPythonPath in the installer looks for
"SOFTWARE\Python\PythonCore\2.7\InstallPath" in the registry

My python install didn't write anything in the windows registry. I think it would be better to look for python in the PATH.

My workaround was to create an registry key with some random string inside to be able to go through getPythonPath

https://github.com/3dtof/voxelsdk/blob/abd9090509c87fd5e5d090dfc291f7b67e0a39e5/config/NSIS.template.in#L902

donjburdette, 2017-10-07:
Hi Arrowana,

I am having the same problem with Voxel.SDK-0.6.10 Setup.  My question is, did your workaround still allow you to use the python binding functionality>

donjburdette, 2017-10-07:
I noticed that I have Python2.7 32 bit version installed.  I uninstalled the 32 bit version and installed the 64 bit version.  I was able to successfully complete the Voxel.SDK-0.6.10 installation.