3dtof/voxelsdk #114: Parameter.h warnings

flugi, 2017-02-20:
There are a lot of warnings in Parameter.h, at least in 0.6.5, and none of them seems to be corrected since. The most interesting is this:

.../include/voxel-0.6.5/Parameter.h:154:7: warning: statement has no effect [-Wunused-value]
       value == _value;
       ^

This seems to be no meaning, however it is like this since the commit "Added support for additive common-phase correction for Calculus in a generic way.", so in 4 days it will have birthday, so maybe just I don't get it.

roshansuresh, 2017-04-11:
Yeah I'm having the same issue as well.

gadiyar, 2017-04-26:
Fixed with commit https://github.com/3dtof/voxelsdk/commit/0f11d7eac5af815b73a767ed6689700fc89a5400

closed 2017-04-26 by gadiyar