3dtof/voxelsdk #115: with/without PCL build would be nice

flugi, 2017-02-20:
I use custom PCL with VTK disabled currently, and I have hard time to hack PCL dependency from VoxelSDK. I only need VoxelSDK as an interface to the camera, I don't want to use point cloud services through it.

A CMakeLists.txt which is able to turn on and off PCL dependent parts of VoxelSDK would be highly appreciated.

larrylisky, 2017-02-20:
All PCL dependency should be confined in the VoxelPCL directory. Remove that library from top level CMakeLists.txt should do it. 

Larry

minghuadev, 2017-05-19:
The App directory also reference PCL. Should that also be removed from the top level CMakeLists.txt too? That's how I'm building it but would like to know other options if somebody has done it.

gadiyar, 2017-06-20:
Looking at using this for rendering in VoxelCLI/SimpleVoxelViewer and getting rid of the PCL dependency:
https://www.opengl.org/resources/libraries/glut/