3dtof/voxelsdk #121: apt-repositories do not have a release file

twelti, 2017-05-04:
When trying to add the apt-repositories for SWIG and PCL, Ubuntu 16.04 LTS refuses to use these repositories:
```
Err:9 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial Release                                         
  404  Not Found
E: The repository 'http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.

```

gadiyar, 2017-05-04:
For Ubuntu 16.04, you should be able to use pcl directly from the default repositories. There may be no need to add this ppa explicitly.

gadiyar, 2017-06-20:
Won't fix as this is not our repository.

closed 2017-06-20 by gadiyar