3dtof/voxelsdk #122: DepthCapture.cpp issue with OPT8241 Tintin

roshansuresh, 2017-05-11:
Hi,

I tried to run the DepthCapture.cpp code in Visual Studio 2013. I linked the requisite libraries (voxel.lib, ti3dtof.lib, voxelpcl.lib and tintincdk.lib) and the include folder for the headers. Visual Studio compiles the code and runs the executable, but immediately shows an error that "The program can't start because voxel.dll is missing from you computer" and in Visual Studio it says that "The program exited with code ....... "A dependent dll file was not found".

So I thought I need to link voxel.dll instead of voxel.lib. I tried that but I get an LNK1107 error saying that the file is either corrupted or invalid.

I don't know what the issue is. Could I get some suggestions to deal with this issue?

Thanks.
[image]
[image]

roshansuresh, 2017-05-12:
I figured out that I need both the .lib and .dll files of voxel. But when I tried that, I got another error from Visual Studio saying that the .dll file is either corrupted or invalid. I'm running VS2013, are the library files compiled for the corresponding VC++ compiler?

gadiyar, 2017-05-15:
You don't need the .lib, just the .dlls files.

Please follow build instructions from https://github.com/3dtof/voxelsdk/wiki/Build-from-source-on-Windows and when you run the binaries, make sure the VOXEL_SDK_PATH and PATH variables are appropriately set. At minimum, you need something like the following:

    set VOXEL_SDK_PATH=C:\Work\voxelsdk\build
    set PATH=%VOXEL_SDK_PATH%\lib;%VOXEL_SDK_PATH%\bin;%PATH%

roshansuresh, 2017-06-01:
Thanks for the reply. But for some reason, I'm not able to download Cypress SuiteUSB from their website. It is one of the dependencies listed in the Build from Source in Windows page. Is there a way to bypass it? Or is there an alternative dependency I can use? 
Also, are the binaries for Voxel SDK supposed to work with the VC++ compiler in Visual Studio 2013 only?

JproChen, 2020-01-20:
@roshansuresh  I have the same problem, did you solved it?