3dtof/voxelsdk #137: Trouble Accessing .vxl data with python.

donjburdette, 2017-10-25:
I am having an issue accessing data from a .vxl file using python.  I have based my code off of portions of the example c++ code: https://github.com/3dtof/voxelsdk/blob/master/App/VXLDecoder.cpp
I am able to use the Voxel.FrameStreamReader and access basic stats of the .vxl file (such as number of frames); however, when I try to parallel the c++ code I have trouble starting at line 125:

 (RawDataFrame *rawFrame = dynamic_cast<RawDataFrame *>(r.frames[DepthCamera::FRAME_RAW_FRAME_UNPROCESSED].get());

Issue: when I try to access reader.frames[DepthCamera.FRAME_RAW_FRAME_UNPROCESSED].get()] in python, I am told that Frames does not have .get() function call.  

My guess is that dynamic casting may not be properly implemented when SWIG generates the python code from the c++ source code...?

Does anyone with experience using the Voxel library in python have experience reading .vxl files and can offer advice?

My code is as follows:

import Voxel

sys = Voxel.CameraSystem()

reader = Voxel.FrameStreamReader('data.vxl', sys)

if(reader.isStreamGood() == False):
      print 'Error: File stream is not good.'
elif(reader.isStreamGood() == True):
      print 'Success: File stream is good'

      numFrames = reader.size()
      print 'There are ' + str(numFrames) + ' data frames in the file.'

      testFrame = reader.frames

      # Attempting to extract frame data
      frame data = reader.frame[0].get()

#-------- Error ------------
Attribute Error: 'Voxel.Frame' object has no attribute 'get'

gadiyar, 2017-10-26:
Could you check if something like the attached python script works for you?

donjburdette, 2017-10-26:
Thank you so much, gadiyar.

It works perfectly.  This is just what I was looking for.

closed 2018-01-09 by suramyag1