3dtof/voxelsdk #138: How to calibrate the OPT8241 camera

Booooooosh, 2017-10-26:
Hi, all:
We are using the OPT8241 camera to fetch the point cloud data. The scene is just about 1-2 meters away form the camera. I looked at the VoxelViewer's instructions and found out that there are so many kinds of calibration, is it enough if I just so the len's distortion calibration?
BTW, there are two ways to do the len's calibration, one is though the VoxelSDK and the second is through the VoxelViewer, are they basically the same? I mean, can I get these distortion parameters form the VoxelViewer and just use them in the VoxelSDK to undistort the point cloud?

Many thanks!

larrylisky, 2017-10-27:
OPT8241-CDK-EVM comes calibrated at "MetrilusLongRange(HW)" profile.  VoxelSDK is a software library enable user program access the underlying 3D-TOF hardware.  Voxel Viewer is primarily a GUI front-end that  calls pre-built VoxelSDK library to accomplish most of its functions.  Lens correction is accomplished by the Voxel/PointCloudTransform class.  When you calibrate a previously uncalibrated camera, you should as a minimum calibrate for lens, common phase offset and pixewise phase offset.

Pacifish, 2018-08-21:
we are Chinese team, and we provide system integration service for OPT8241. Tech discuss is always appreciate.

suramyag1, 2018-08-21:
Hi, 

Please contact ti-3dtof@list.ti.com for discussions.