3dtof/voxelsdk #139: How to temporary apply camera lens calibration parameters and then reset it back to defaults?

vicrucann, 2017-10-27:
I have an instance of `DepthCamera` (TinTinCDK) and after performing the camera lens calibration (to obtain focal length, principle point, etc), I'd like to test the calibration parameters by trying to fit the resulting point cloud to a plane. 

**The question is:** how to temporary apply the calibration parameters (fx, fy, px, py, distortion params) so that to obtain an undistorted point cloud? After that I'd like to reset the parameters back to defaults. Is it even possible? 

Looking at the `DepthCamera` API, I can see different methods like `addFilter()` or `addCameraProfile()`. Would it be one of this methods? Or something else? 

Thanks!

larrylisky, 2017-10-27:
Victoria

Are you trying to use a different lens calibration or no calibration?  

One way they do what you are asking is to create a new calibration profile from a desired parent and then perform only lens calibration on it. Then edit the lens parameter to whatever you want. 

The new profile configuration file could be found under .Voxel/conf of your home directory. 

-Larry

vicrucann, 2017-10-27:
Larry, 

I'm doing my own calibration. I already got all the coefficients: fx, fy, px, py and lens distortion and want to replace what is used by default. Now I would like to use the calibration coefficients in order to obtain an undistorted cloud. You mention that I should try creating a new calibration profile from a parent. Will I be able to revert the changes back to defaults? If yes, could you tell what methods to use for that? 

Currently I'm thinking to use:

```
int id = depthCamera->addCameraProfile("MyLensOnly", -1);
Voxel::ConfigurationFile* configFile = depthCamera->configFile.getCameraProfile(id);
configFile->setFloat("calib", "fx", intrinsics.fx);
// set other lens params

configFile->write(); // is this correct to use?

// ... and now if I want to restore the default profile, what's the best way to do it?
// ...
```
Does this sound right?

Btw, thanks for the fast reply!

larrylisky, 2017-10-27:
Well, since the new profile that you create is inherited from a  parent profile, to go back to "default", just select the parent profile instead of your new one.

larrylisky, 2017-10-27:
It is probably easier to create a new profile using Voxel Viewer, and then edit it by hand.  Voxel SDK will read from the .Voxel/conf directory to retrieve all the profiles, then your program selects which profile to load onto the hardware.  Creating and adding new profile programmatically is not the way to go, IMO.

vicrucann, 2017-10-27:
I see, thanks for clarification. Unfortunately, for us, we would need to somehow change the lens parameters programmatically, on the go and then revert them back. 

I was looking into the `DepthCamera` interface and found it has an instance of `PointCloudFrameGenerator` which uses `PointCloudTransform`. I noticed that transform contains the parameters that are potentially what I need to change. Do you think it is a way to try to subclass `DepthCamera` and set the lens parameters through using the generator and transform to what I need, and then reset them back? In your opinion Is it meaningful / doable?

larrylisky, 2017-10-27:
If you have to change configuration parameters, such as lens parameters, the best way is probably load in the "default" configuration and programmatically substitute the lens parameters with your own.  Example of how to read and write configuration parameters can be found under Test/CameraSystemConfiguration.cpp, the parameter write is at this [line](https://github.com/3dtof/voxelsdk/blob/master/Test/CameraSystemConfigTest.cpp#L346). 

To return to default, just reload the configuration.  Example is [here](https://github.com/3dtof/voxelsdk/blob/master/Test/CameraSystemConfigTest.cpp#L346).

vicrucann, 2017-10-27:
Larry, thanks so much for help! I will be looking at those examples; this is probably what I need.

closed 2017-10-27 by vicrucann