3dtof/voxelsdk #147: Add the ability to save point cloud frames in the .pcd format

suramyag1, 2018-03-26:
VoxelSDK saves point clouds in a format different to the standard .pcd

Add the ability to save the point cloud frame using the *.pcd format.

YonseiLidar, 2019-01-22:
Please share how you did make it. Here still I got the same problem to match the point clouds data format when saving.

suramyag1, 2019-01-22:
@YonseiLidar This hasn't been done yet. However, I do have a python script to save the point cloud data to the pcd format (note that this is saved as text, so the file size is quite large, even for a single frame). Replace txt with py. You need python 2.7 for this, with numpy and tqdm along with Voxel libraries built for python. 

You can do something similar in C++
[pointCloudExtract.txt](https://github.com/3dtof/voxelsdk/files/2782742/pointCloudExtract.txt)

Hope this helps. 

Suramya