3dtof/voxelsdk #149: SimpleVoxelViewer crash when addCoordinateSystem is called

cwlmyjm, 2018-05-16:
I now have a OPT9221 sensor and want to review the example code to start my programming. But when I run SimpleVoxelViewer, the program always crash. I try to debug the program with RelWithDebInfo mode and found it always crash when addCoordinateSystem is called. Then I download pcl project, set a breakpoint in addCoordinateSystem, attach to SimpleVoxelViewer process and the breakpoint also was hited. That means addCoordinateSystem was called correctly, but I found the value of id become "e" instead the default "reference". Function addCoordinateSystem was called with single argument, so the value of id should be set to the default value but it does not. To find which project lead that problem, I create a project with PCL library and usee some method to see the value of id, but this give the value correctly. So I am confused with it.
[image]
I am running my program on Win10 x64 with QT 5.4.0, PCL 1.7.2 with VS 2013.

cwlmyjm, 2018-05-17:
Windows got following error message

It shows the program throw an exception in pcl_visualization_debug.dll, I think there should have problem in config but I am not sure where.

线程 0x38e8 已退出，返回值为 0 (0x0)。
“SimpleVoxelViewer.exe”(Win32):  已卸载“C:\Windows\System32\ResourcePolicyClient.dll”
“SimpleVoxelViewer.exe”(Win32):  已加载“C:\Windows\System32\DriverStore\FileRepository\igdlh64.inf_amd64_2681fc9dedd95cd9\igc64.dll”。用户取消了自动符号加载。
0x00007FFF8DB6390E (pcl_visualization_debug.dll) (SimpleVoxelViewer.exe 中)处的第一机会异常:  0xC0000005:  读取位置 0xFFFFFFFFFFFFFFFF 时发生访问冲突。
0x00007FFF8DB6390E (pcl_visualization_debug.dll) (SimpleVoxelViewer.exe 中)处有未经处理的异常:  0xC0000005:  读取位置 0xFFFFFFFFFFFFFFFF 时发生访问冲突。

>	pcl_visualization_debug.dll!std::_Iterator_base12::_Adopt(const std::_Container_base12 * _Parent) 行 165	C++
 	pcl_visualization_debug.dll!std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > >::_String_const_iterator<std::_String_val<std::_Simple_types<char> > >(const char * _Parg, const std::_Container_base12 * _Pstring) 行 54	C++
 	pcl_visualization_debug.dll!std::basic_string<char,std::char_traits<char>,std::allocator<char> >::end() 行 1615	C++
 	pcl_visualization_debug.dll!boost::hash_value<char,std::allocator<char> >(const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & v) 行 386	C++
 	pcl_visualization_debug.dll!boost::hash<std::basic_string<char,std::char_traits<char>,std::allocator<char> > >::operator()(const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & v) 行 456	C++
 	pcl_visualization_debug.dll!boost::unordered::detail::mix64_policy<unsigned __int64>::apply_hash<boost::hash<std::basic_string<char,std::char_traits<char>,std::allocator<char> > >,std::basic_string<char,std::char_traits<char>,std::allocator<char> > >(const boost::hash<std::basic_string<char,std::char_traits<char>,std::allocator<char> > > & hf, const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & x) 行 620	C++
 	pcl_visualization_debug.dll!boost::unordered::detail::table<boost::unordered::detail::map<std::allocator<std::pair<std::basic_string<char,std::char_traits<char>,std::allocator<char> > const ,vtkSmartPointer<vtkProp> > >,std::basic_string<char,std::char_traits<char>,std::allocator<char> >,vtkSmartPointer<vtkProp>,boost::hash<std::basic_string<char,std::char_traits<char>,std::allocator<char> > >,std::equal_to<std::basic_string<char,std::char_traits<char>,std::allocator<char> > > > >::hash(const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & k) 行 767	C++
 	pcl_visualization_debug.dll!boost::unordered::detail::table_impl<boost::unordered::detail::map<std::allocator<std::pair<std::basic_string<char,std::char_traits<char>,std::allocator<char> > const ,vtkSmartPointer<vtkProp> > >,std::basic_string<char,std::char_traits<char>,std::allocator<char> >,vtkSmartPointer<vtkProp>,boost::hash<std::basic_string<char,std::char_traits<char>,std::allocator<char> > >,std::equal_to<std::basic_string<char,std::char_traits<char>,std::allocator<char> > > > >::operator[](const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & k) 行 345	C++
 	pcl_visualization_debug.dll!boost::unordered::unordered_map<std::basic_string<char,std::char_traits<char>,std::allocator<char> >,vtkSmartPointer<vtkProp>,boost::hash<std::basic_string<char,std::char_traits<char>,std::allocator<char> > >,std::equal_to<std::basic_string<char,std::char_traits<char>,std::allocator<char> > >,std::allocator<std::pair<std::basic_string<char,std::char_traits<char>,std::allocator<char> > const ,vtkSmartPointer<vtkProp> > > >::operator[](const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & k) 行 1190	C++
 	pcl_visualization_debug.dll!pcl::visualization::PCLVisualizer::addCoordinateSystem(double scale, const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & id, int viewport) 行 594	C++
 	SimpleVoxelViewer.exe!Voxel::PCLViewer::_renderLoop() 行 30	C++
 	[外部代码]	

[image]

gadiyar, 2018-05-17:
If you're using RelWithDebInfo build, then your PCL should also be the release version not the debug version.

Could you please check if you are using the Debug version of PCL? If so, please try with the release version.

cwlmyjm, 2018-05-18:
After switch pcl library to release version, the program call addCoordinateSystem without expection. But it still not work, the pcl_visualization window only display one frame (I think it is one frame) but crash  immediately. Here is a screenshot:

[image]

To find out which function leads the problem, I download vtk library, generate debug information and add these to debugger. Here is the call stack:

>	msvcr120d.dll!_CrtDbgBreak() 行 87	C
 	msvcr120d.dll!_VCrtDbgReportW(int nRptType, void * returnAddress, const wchar_t * szFile, int nLine, const wchar_t * szModule, const wchar_t * szFormat, char * arglist) 行 506	C
 	msvcr120d.dll!_CrtDbgReportWV(int nRptType, void * returnAddress, const wchar_t * szFile, int nLine, const wchar_t * szModule, const wchar_t * szFormat, char * arglist) 行 263	C++
 	msvcr120d.dll!_CrtDbgReportW(int nRptType, const wchar_t * szFile, int nLine, const wchar_t * szModule, const wchar_t * szFormat, ...) 行 279	C++
 	msvcr120d.dll!_free_dbg_nolock(void * pUserData, int nBlockUse) 行 1328	C++
 	msvcr120d.dll!_free_dbg(void * pUserData, int nBlockUse) 行 1265	C++
 	msvcr120d.dll!free(void * pUserData) 行 50	C++
 	vtkCommonCore-6.1.dll!vtkDataArrayTemplate<float>::DeleteArray() 行 220	C++
 	vtkCommonCore-6.1.dll!vtkDataArrayTemplate<float>::~vtkDataArrayTemplate<float>() 行 92	C++
 	[外部代码]	
 	vtkCommonCore-6.1.dll!vtkObjectBase::UnRegisterInternal(vtkObjectBase * __formal, int check) 行 232	C++
 	vtkCommonCore-6.1.dll!vtkObject::UnRegisterInternal(vtkObjectBase * o, int check) 行 905	C++
 	vtkCommonCore-6.1.dll!vtkObjectBase::UnRegister(vtkObjectBase * o) 行 190	C++
 	vtkCommonDataModel-6.1.dll!vtkFieldData::InitializeFields() 行 219	C++
 	vtkCommonDataModel-6.1.dll!vtkDataSetAttributes::InitializeFields() 行 239	C++
 	vtkCommonDataModel-6.1.dll!vtkFieldData::Initialize() 行 237	C++
 	vtkCommonDataModel-6.1.dll!vtkDataSetAttributes::Initialize() 行 267	C++
 	vtkCommonDataModel-6.1.dll!vtkDataSetAttributes::ShallowCopy(vtkFieldData * fd) 行 193	C++
 	vtkCommonDataModel-6.1.dll!vtkDataSet::ShallowCopy(vtkDataObject * dataObject) 行 348	C++
 	vtkCommonDataModel-6.1.dll!vtkPointSet::ShallowCopy(vtkDataObject * dataObject) 行 388	C++
 	vtkCommonDataModel-6.1.dll!vtkPolyData::ShallowCopy(vtkDataObject * dataObject) 行 1813	C++
 	vtkRenderingCore-6.1.dll!vtkPrimitivePainter::PrepareForRendering(vtkRenderer * renderer, vtkActor * actor) 行 123	C++
 	vtkRenderingCore-6.1.dll!vtkPainter::Render(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 212	C++
 	vtkRenderingCore-6.1.dll!vtkPolyDataPainter::Render(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 62	C++
 	vtkRenderingCore-6.1.dll!vtkChooserPainter::RenderInternal(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 319	C++
 	vtkRenderingCore-6.1.dll!vtkPainter::Render(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 213	C++
 	vtkRenderingCore-6.1.dll!vtkPolyDataPainter::Render(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 62	C++
 	vtkRenderingCore-6.1.dll!vtkPainter::RenderInternal(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 224	C++
 	vtkRenderingOpenGL-6.1.dll!vtkOpenGLCoincidentTopologyResolutionPainter::RenderInternal(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 56	C++
 	vtkRenderingCore-6.1.dll!vtkPainter::Render(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 213	C++
 	vtkRenderingCore-6.1.dll!vtkPolyDataPainter::Render(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 62	C++
 	vtkRenderingCore-6.1.dll!vtkPainter::RenderInternal(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 224	C++
 	vtkRenderingOpenGL-6.1.dll!vtkOpenGLRepresentationPainter::RenderInternal(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 86	C++
 	vtkRenderingCore-6.1.dll!vtkPainter::Render(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 213	C++
 	vtkRenderingCore-6.1.dll!vtkPolyDataPainter::Render(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 62	C++
 	vtkRenderingCore-6.1.dll!vtkPainter::RenderInternal(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 224	C++
 	vtkRenderingOpenGL-6.1.dll!vtkOpenGLLightingPainter::RenderInternal(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 118	C++
 	vtkRenderingCore-6.1.dll!vtkPainter::Render(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 213	C++
 	vtkRenderingCore-6.1.dll!vtkPolyDataPainter::Render(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 62	C++
 	vtkRenderingCore-6.1.dll!vtkPainter::RenderInternal(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 224	C++
 	vtkRenderingCore-6.1.dll!vtkCompositePainter::RenderInternal(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 70	C++
 	vtkRenderingCore-6.1.dll!vtkPainter::Render(vtkRenderer * renderer, vtkActor * actor, unsigned long typeflags, bool forceCompileOnly) 行 213	C++
[... cut ...]
 	vtkRenderingOpenGL-6.1.dll!vtkWin32RenderWindowInteractor::OnTimer(HWND__ * __formal, unsigned int timerId) 行 554	C++
 	vtkRenderingOpenGL-6.1.dll!vtkHandleMessage2(HWND__ * hWnd, unsigned int uMsg, unsigned __int64 wParam, __int64 lParam, vtkWin32RenderWindowInteractor * me) 行 826	C++
 	vtkRenderingOpenGL-6.1.dll!vtkHandleMessage(HWND__ * hWnd, unsigned int uMsg, unsigned __int64 wParam, __int64 lParam) 行 707	C++
 	[外部代码]	
 	vtkRenderingOpenGL-6.1.dll!vtkWin32RenderWindowInteractor::Start() 行 129	C++
 	[外部代码]	
 	SimpleVoxelViewer.exe!Voxel::PCLViewer::_renderLoop() 行 76	C++
 	[外部代码]