3dtof/voxelsdk #154: Missing deb package in libvoxel-0.6.10-amd64-xenial-all.tgz

robust-lu, 2018-06-21:
In libvoxel-dev-0.6.9-amd64-xenial.deb, it seems no libvoxel-dev-0.6.10. is there anything wrong?

suramyag1, 2018-07-03:
libvoxel-dev-0.6.9-amd64-xenial.deb is for version 0.6.9, while libvoxel-dev-0.6.10-amd64-xenial.deb is for 0.6.10. 

0.6.9 will not have contents for 0.6.10, as this is an older version. 

Could you explain the problem further?

dknyxh, 2018-12-20:
Hi, 
I think what he means to say is that in the current release of `libvoxel-0.6.10-amd64-xenial-all.tgz` file, its missing the package : `libvoxel-dev-0.6.10`. This file is present in `0.6.9` but not in `0.6.10`. This will cause the installation of prebuilt libraries to fail.

robust-lu, 2018-12-25:
Yes, @dknyxh said is what i mean. The 0.6.10 package don't have libvoxel-dev-0.6.10-amd64-xenial.deb and the 0.6.10 installation of prebuilt libraries to fail. How to solve it?

closed 2019-03-08 by robust-lu