3dtof/voxelsdk #156: WARNING: CameraSystem: No depth camera library found or loaded.

pans0ul, 2018-07-06:
Ive built project and could run the VoxelCLI.. with:  https://github.com/3dtof/voxelsdk/wiki/Installation-on-Linux

My os is Ubuntu 14.04LTS ,I ran VoxelCLI ,but it can not connete the depth camera .why? 
if the problem is No depth camera library found or loaded, How do I solve this problem? 
please help me ,thanks a lot!

the picture is as follow:
[image]

pans0ul, 2018-07-09:
please help me

suramyag1, 2018-07-09:
Hi! 

Have you updated the PATH variables correctly? Use these variables if you've built voxelsdk:
https://github.com/3dtof/voxelsdk/wiki/Build-from-source-on-Linux#running-from-source

suramyag1, 2018-07-12:
@pans0ul 

Did this resolve the issue? Can I close this thread?

pans0ul, 2018-07-12:
@suramyag1   sure, thanks!

closed 2018-07-12 by suramyag1