3dtof/voxelsdk #157: APPCRASH when voxelsdk-examples DepthCapture was excuted in win7

jarodlee0304, 2018-07-16:
Hello,
I excuted the voxelsdk-examples DepthCapture in win7 and App was crashed.
Software
OS: win7 Service Pack 1
Cmake (cmake-gui) Version: 3.12.0
PCL-1.7.2-AllInOne-msvc2015-win64
Visual studio 14 2015 Win64
DemoApplications: DepthCapture
Hardware
OPT8241-CDK-EVM
When I used cmake-gui to build project, Configuring and Generating were both done.
Command in Windows terminal: cmake --build . --target ALL_BUILD --config Release, no error but 126 warning.
Executed App DepthCapture in win7 terminal: DepthCapture.exe -v 0x0451 -p 0x9105 -s 12019461200050 -f filename.bin -n 100 -t raw_processed

suramyag1, 2018-07-16:
Hi @jarodlee0304, 

It seems that you built a new version of the SDK, but the VOXEL_SDK_PATH is still set to D:\Program Files (x86)\Voxel SDK .. 
Please change that and try running it again. 

Suramya

jarodlee0304, 2018-07-17:
Hi Suramya,
I installed the Voxel.SDK-0.6.10-win64.exe in win7 OS directly, not built SDK. So VOXEL_SDK_PATH was set to D:\Program Files (x86)\Voxel SDK 0.6.10

suramyag1, 2018-07-17:
@jarodlee0304 

I'm sorry, i thought you built VoxelSDK as well. This could be a compatibility issue, then. VoxelSDK was built using VS2013. 
If you want to use VS2015, you may need to build the SDK yourself.

jarodlee0304, 2018-07-18:
Hi Suramya,
When I installed the release version Voxel.SDK-0.6.10-win64.exe in win7 OS, chose PCL 1.7.2 and Qt 5.8 winrt_x64_msvc2015 for it already.

suramyag1, 2018-07-18:
@jarodlee0304 

The release has been compiled using VS2013, so specifying msvc2015 will not work. 

If you want a VS2015 release, please use the older release (https://github.com/3dtof/voxelsdk/releases/download/v0.6.9/Voxel.SDK-0.6.9-msvc2015-win64.exe)

closed 2018-07-19 by jarodlee0304