3dtof/voxelsdk #160: Issue of "Build from voxelsdk 0.6.11 source on Win10 64-bit" process

CJHsiehinleda, 2018-07-26:
Hi
I encountered a compiled error during building process.
Error message during compile with VS2015 64-bit and voxelsdk 0.6.11 (from releases site) is shown below
[image]
How to solve it? 

Before compiling, I installed
1. PCL 1.8.0
2. Doxygen 1.8.9.1
3. swingwin 3.0.12
4. graphviz  2.38
5. Qt 5.6.3 for 64-bit
And set PATH as
[image]
Do I get the wrong configuration?

suramyag1, 2018-07-27:
Hi @CJHsiehinleda 

Can you try building in the Release mode? 

It seems that you already have Voxel SDK 0.6.10 installed. I don't think this is needed to build 0.6.11 (you can remove those lines from the path, although it wouldn't make a difference). 

Suramya

CJHsiehinleda, 2018-08-02:
After switch to Release mode, I finally can compile. 
Thanks! @suramyag1

closed 2018-08-02 by CJHsiehinleda